z88dk/z88dk #720: build: problems with ext/uthash submodule

aralbrec, 2018-04-29:
While trying to update ext/uthash I get this error:

```
git -c diff.mnemonicprefix=false -c core.quotepath=false submodule update --init
error: no such remote ref 29fc26a8ee4116236230160b67ae693a11a9352f
Fetched in submodule path 'ext/uthash', but it did not contain 29fc26a8ee4116236230160b67ae693a11a9352f. Direct fetching of that commit failed.
Completed with errors, see above.
```

The commit `29fc26a8ee4116236230160b67ae693a11a9352f` is not in z88dk/uthash but is in the upstream https://github.com/troydhanson/uthash at a later date.

So I think z88dk/uthash has to be updated to grab this later commit from upstream.  I'm not that familiar with git flow so I haven't attempted this.

pauloscustodio, 2018-04-29:
You can try to delete the ext/uthash directory and call ```git submodule update --init --recursive``` to get it back from the repository.

aralbrec, 2018-04-30:
Deleting ext/uthash and refetching gets the same result:

```
alvin@WINDOWS-O85KGHF MINGW32 /c/Projects/z80/z88dk/z88dk (master)
$ git submodule update --init --recursive
error: no such remote ref 29fc26a8ee4116236230160b67ae693a11a9352f
Fetched in submodule path 'ext/uthash', but it did not contain 29fc26a8ee4116236230160b67ae693a11a9352f. Direct fetching of that commit failed.

alvin@WINDOWS-O85KGHF MINGW32 /c/Projects/z80/z88dk/z88dk (master)
$ 
```

I think the error is legitimate.  Our version `z88dk/uthash` comes from Sep 7, 2017, but the commit referred to `29fc26a8ee4116236230160b67ae693a11a9352f` is only available in upstream [troydhanson/uthash](https://github.com/troydhanson/uthash/commits/master) on Feb 15, 2018.  I'm wondering if you're using troydhanson/uthash instead of z88dk/uthash?

aralbrec, 2018-04-30:
Ok I can see in [.gotmodules](https://github.com/z88dk/z88dk/blob/master/.gitmodules) that a few days ago we changed away from z88dk/uthash to troydhanson/uthash and maybe my git frontend is not paying attention to this file.

```
[submodule "UNIXem"]
	path = ext/UNIXem
	url = https://github.com/z88dk/UNIXem.git
[submodule "ext/optparse"]
	path = ext/optparse
	url = https://github.com/skeeto/optparse.git
[submodule "ext/regex"]
	path = ext/regex
	url = https://github.com/geoffmcl/regex.git
[submodule "ext/uthash"]
	path = ext/uthash
	url = https://github.com/troydhanson/uthash.git
[submodule "ext/Unity"]
	path = ext/Unity
	url = https://github.com/ThrowTheSwitch/Unity.git
```

We have forks of some of these submodules in z88dk/Unity, z88dk/uthash, z88dk/Unixem but it appears we're only using z88dk/Unixem and have gotten rid of the other forks?  Maybe we want to update our forks instead?

pauloscustodio, 2018-04-30:
The only fork where we had to commit anything was UNIXem. The others where just copies and not up-to-date, so I thought it was best to refer to the original repo instead of the fork. 
You may try to remove these entries from .git/config and re-inir the submodules.

pauloscustodio, 2018-04-30:
I don't know if it is best to fork and update the submodules or to clone the original. Any more experienced git user can answer?

pauloscustodio, 2018-04-30:
https://blog.github.com/2016-02-01-working-with-submodules/

aralbrec, 2018-05-01:
Another link, search for "pulling in upstream changes":
https://git-scm.com/book/en/v2/Git-Tools-Submodules

Submodules do what I didn't think they did.  When you import as a submodule, you're grabbing a snapshot at a specific commit so upstream changes do not propagate to the submodule.  You have to manually merge from upstream to update the submodule if that's what you want to do.  You can see our [submodules](https://github.com/z88dk/z88dk/tree/master/ext) and the commit they're based on.

I think if we want to change the code, I'd use the fork to hold our changes and merge upstream changes into that.  Then import our fork as the submodule.

Anyway, I manually changed the submodule source for ext/uthash and the problem is solved.

closed 2018-05-01 by aralbrec