z88dk/z88dk #722: [ZX80][info] gen_tv_field

Fabrizio-Caruso, 2018-04-29:
I am trying to prevent the screen from rolling during gameplay with my game.

I do not understand how I should use gen_tv_field.
If I put more I get sometimes worst results. I can minor improvemnts but randomly inserting 
or removing gen_tv_field in my code.

Is there a better approach?

zx70, 2018-05-02:
not much to say, you need to have the tv frame generation happening frequently..  it is very challenging in big programs like yours and long time consuming operations (even a single slow function call) must be avoided.  I think that a minimun amount of time must be spent before syncing again, but you don't need to be too accurate, for what I could observe.
You may experiment with short loops first and then extend them gradually when you have a stable picture...  to make dstar.c running smoothly wasn't easy at all, you might start by tweaking it and looking the way it changes with little program mods.

Fabrizio-Caruso, 2018-05-02:
What I tried is to but gen_tv_field after evert non-trivial operation but I may have skipped one or two and the result was horrible. On the ZX80, I have set my game in turn-based mode only, I thought this would allow me to make the movements smoother but it has turned out to be much harder than I expected.

closed 2018-05-02 by Fabrizio-Caruso

zx70, 2018-05-03:
I think dstar.c is the first game written in C (and not in assembly) running correctly with a stable picture on the EightyOne emulator (never tester on the real hw I think).
As I said it is not an easy task,  but it is possible that you see a worse picture while calibrating your loops, even if you just inserted a necessary extra sync call