z88dk/z88dk #724: (vg5k) Should be compiled with ixiy

suborb, 2018-04-30:
Fiddling around trying to create the generic console for the vg5k and running into a few stability issues.

Disassembling the ROM shows that the interrupt uses ix, so we should compile it with --ixiy so that nothing touches ix. This seems to improve stability and makes things like adv_a playable without crashing (printf via stdio uses ix hence the problem).

I've run into the same problem as @Fabrizio-Caruso in #308 when poking the video memory directly - it doesn't refresh - BASIC is obviously doing something to flush out the display so I'll need to find out what that is.

suborb, 2018-04-30:
Not relevant, but...:

```
ld a,n
out (0xcf),a
```

Changes the border. Where n is one of the usual VG5k colours.

Fabrizio-Caruso, 2018-04-30:
I have observed a problem, when I try to redefine characters with something like:
.org $6000

           ld hl,_graphic
           ld a,33 ; code caractere" ! "
           call $001B
_boucle:
           ld hl,_EF9345
           call $00AD
           ret
           
_graphic:
.db 255,0,255,0,255,0,255,0,255,0
  
_EF9345:   
.db 28,$20,$81,$29,00,$20,$82,$29,$00,$20,$83,$29,$67,$20,$84,$29,$13,$21,33,$22,$80,$23,22,$26,20,$27,20,$28,00

When I return, the redefinition is lost and I get random behavior.

If I stay within Assembly, I can see the character correcrly redefined.

Fabrizio-Caruso, 2018-05-01:
For some strange reasons my game runs stable after playing quite some time.

suborb, 2018-05-01:
The generic terminal is working well now that -DSTANDARDESCAPECHARS has been enabled for the vg5k - still a bit sluggish but that can be improved.

I'm trying to get the ansiterminal working with interrupts enabled but it's proving elusive to say the least:

[image]

 For some reason sp ends up pointing to ROM which is a little odd...

Fabrizio-Caruso, 2018-05-01:
@suborb, I will try to find the doc (in French) that I have seen some time ago.
My (limited) understanding of VG5K's EF9345 is that is not using memory mapped memory even though within BASIC it appears so. 
You use EF9345 by sending data and commands (somehow similarly to the VDC) by calling $00AD$ with HL pointing at sequences of commands and data. The rom routine at $00AD$ is probably writing in some data/control ports as the VDC.

If my understanding is correct, it should be treated as a memory mapped system.
 
The example I showed above is doing multiple things (to draw '!' at a given position in two-color mode):
$21,33 should stand for the code 33=" ! " 
$22,80 sets two-color mode;
$23,22 set fore/background colors
$26 is the y position (range 7 - 31 to mean 0 - 25) 
$27 is the x position (0 à 39)

Fabrizio-Caruso, 2018-05-01:
A more usable link is
http://vg5000.free.fr/docs/livres/Clefs%20pour%20VG5000.pdf

suborb, 2018-05-01:
I agree, I think the memory region is just there for use as a scroll buffer! Certainly sending all 2000 bytes out via out is very slow so I'm now just using it for cls/scrolling.

If I can manage to download that file (I think my registration has failed?) then addressing the VDC directly using those sort of commands could be an interesting way to support more than just characters as well as the other screen modes that the chip supports, but the vg5k doesn't actually use.

Fabrizio-Caruso, 2018-05-01:
@suborb Use the PDF link I have added. I have managed to download it.

Fabrizio-Caruso, 2018-05-01:
The VG5k can, sort of, do multicolor graphics by using multi-color redefined characters. You can make colored software sprites (bu using multicolorUDG) and some people have already done it.
If I am not wrong the EF9345 is used in the TRS-80 MC-10 and its clone the Alice Matra 4k and 16k.

suborb, 2018-05-01:
That doc just refers me to the Thomson documentation for the EF9345! 

I'll look at the Mame driver source and decode the datasheet to figure out what can be done.

Fabrizio-Caruso, 2018-05-01:
@suborb page 77 says something about $00AD. I once had a nice diagram that explained the commands you could send to the routine at $00AD. I will try to find it.

Fabrizio-Caruso, 2018-05-01:
[image]

Fabrizio-Caruso, 2018-05-01:
@suborb, On FB, you can find a very active developer (https://www.facebook.com/profile.php?id=100014125119648) who has managed to understand how the EF9345 chip works. He may help.

Fabrizio-Caruso, 2018-05-01:
@suborb, by the way the EF9345 chip is also used in Minitel! It has an 80 col mode. Lot at the diagram above, to see what it can do.

Fabrizio-Caruso, 2018-05-02:
A more detailed diagrams (done by https://www.facebook.com/profile.php?id=100014125119648):
[image]

Fabrizio-Caruso, 2018-05-02:
@suborb One more example with an updated diagram from the FB VG-5000 guy:
[image]

Fabrizio-Caruso, 2018-05-02:
@suborb 
A description of the rom may also help
[vg5000-rom.txt](https://github.com/z88dk/z88dk/files/1967474/vg5000-rom.txt)

suborb, 2018-05-02:
@Fabrizio-Caruso thanks for all this information - I'm hoping we've got enough to be able to do something now! 

I think for you, redefining characters and displaying them would be good?

BTW, this root issue (now merged) means that you shouldn't need to set ix explicitly now before calling ROM code. Though if you do need to, you'll need to refer to ix as iy, so your code will be ld iy,$47FA

suborb, 2018-05-02:
Closing this issue since the root problem was merged in #725 

EF9345 enhancements will be in a new issue.

closed 2018-05-02 by suborb

Fabrizio-Caruso, 2018-05-02:
Thanks @suborb! Redefining characters and being able to display them would be great!
There are weird things with the VG-5K, some time ago I found how to hide the cursor:
void no_cursor(void)
{	
	#asm
	jr clean_cursor
	ef9345:
		defb 0x04,0x20,0x82,0x29,0x00
	clean_cursor:
		ld hl,ef9345
		;ld ix,$47FA
		call 0x00AD
	#endasm		
}

which, by some unknown reason, only just works after being used and after some display operations the cursor reappears again (as if the bios were re-enabling it).
My solution is to keep on calling the no_cursor routine.

I have observed something similar with UDG definition. I need to re-test these things to be sure.

Fabrizio-Caruso, 2018-05-02:
@suborb, @zx70 
If I remember well I talked about this solution to @zx70 who has implemented something like the no_cursor routine. Maybe it is called all the time in Z88DK's lib. Maybe we are doing something wrong.

Again, I have a limited understanding of Assembly and even less of the VG5000. It looks like something is reverting things (the bios?).