z88dk/z88dk #729: Z180 - use of undocumented Z80 instructions in stdio & stdlib

feilipu, 2018-05-04:
Chasing some failures in the use of the stdio library with YAZ180, I noted that [`asm_fflush.asm`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/stdio/z80/asm_fflush.asm#L51) uses some undocumented Z80 op-codes, that aren't supported on the Z180. Ouch.

A quick `rgrep` shows this was not a solitary case.
```
phillip:~/Z80/z88dk/libsrc/_DEVELOPMENT$ rgrep ixl stdio/
stdio/z80/asm_fflush.asm:   ld a,ixl
stdio/z80/asm_fflush_unlocked.asm:   ld a,ixl

phillip:~/Z80/z88dk/libsrc/_DEVELOPMENT$ rgrep ixl stdlib/
stdlib/z80/__strtoul__.asm:   ld ixl,d
stdlib/z80/__strtoul__.asm:   cp ixl                      ; digit in 0..base-1 ?
stdlib/z80/__strtoul__.asm:   ld a,ixl                    ; a = base
stdlib/z80/__strtoul__.asm:   ld c,ixl                    ; c = base
stdlib/z80/__strtoull__.asm:   ld ixl,d
stdlib/z80/__strtoull__.asm:   cp ixl                      ; digit in [0,base-1] ?
stdlib/z80/__strtoull__.asm:   ld a,ixl                    ; a = base
stdlib/z80/__strtoull__.asm:   ld c,ixl                    ; c = base
stdlib/z80/asm_lltoa.asm:   or ixl
stdlib/z80/asm_ltoa.asm:   or ixl
stdlib/z80/asm_ulltoa.asm:   or ixl
stdlib/z80/asm_ultoa.asm:   or ixl

phillip:~/Z80/z88dk/libsrc/_DEVELOPMENT$ rgrep ixh stdio/
stdio/z80/asm_fflush.asm:   or ixh
stdio/z80/asm_fflush_unlocked.asm:   or ixh

phillip:~/Z80/z88dk/libsrc/_DEVELOPMENT$ rgrep ixh stdlib/
stdlib/z80/asm_lltoa.asm:   ld a,ixh
stdlib/z80/asm_ltoa.asm:   ld a,ixh                    ; check for NULL buf
stdlib/z80/asm_ulltoa.asm:   ld a,ixh                    ; check for NULL buf
stdlib/z80/asm_ultoa.asm:   ld a,ixh                    ; check for NULL buf
```
Could we fix them in the standard libraries, please?
Not sure I could do the best job, but let me know if needed.

aralbrec, 2018-05-04:
The assembler is supposed to generate errors for all the undocumented instructions when in z180 mode.  I'll fix that one and open a new issue for z80asm.

aralbrec, 2018-05-04:
There is no error in the library itself:

[code]
Object  file z180.lib at $134FB8: Z80RMF11
  Name: asm_fflush
  Section code_clib: 0 bytes
  Section code_stdio: 0 bytes
  Symbols:
    L C $0000 __CLIB_OPT_MULTITHREAD (section "") (file config_private.inc:143)
    L C $0000 __CLIB_OPT_STDIO (section "") (file config_private.inc:205)
    G = $0000 asm_fflush (section code_stdio) (file stdio/z80/asm_fflush.asm:86)
  Externs:
    U         asm_fflush_unlocked
  Expressions:
    E =  $0000 $0000: asm_fflush := asm_fflush_unlocked (section code_stdio) (file stdio/z80/asm_fflush.asm:86)
[/code]

This is because "asm_fflush" is being aliased to "asm_fflush_unlocked", which happens when the [multi-threading option](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/stdio/z80/asm_fflush.asm#L23) is off (and it must be off because it's not implemented!).

I verified that z80asm is catching the specific undocumented instructions in `asm_fflush.asm` using this test file:

zzz.asm
```
ld a,ixh
or ixl
ret
```

z80asm -b zzz.asm
assembles fine because it's in z80 mode

z80asm -b --cpu=z180 zzz.asm
generates errors:

```
Error at file 'zzz.asm' line 1: illegal identifier
Error at file 'zzz.asm' line 2: illegal identifier
2 errors occurred during assembly
```

So z80asm will catch any errors while assembling the library.

feilipu, 2018-05-04:
Yes, this is not the issue I'm tracking down. Just noticed it in the process of elimination of other things.
(Unrelated code that used to work, that magically doesn't any more).

That this is aliased away is / was lucky. So the issue would only appear if multi-threaded Z180 library code were assembled, and then the assembler would have called out the error, immediately drawing attention.

commit 21601be1 (2018-05-04): Fix #729:  Replace instance of z80-only code with z180 code for non-z80 targets
closed 2018-05-04 by aralbrec