z88dk/z88dk #731: [udg][feature] UDG support in VG5K and ACE

Fabrizio-Caruso, 2018-05-04:
It would be cool to have support for user-defined graphics in VG5K and ACE.

suborb, 2018-05-05:
I’ve added support for changing the font on ace as well as being able to define up to 32 udgs. 

Sys/ioctl.h has the controls to define then. They overwrite the block graphics. 

In theory the same code will work on vdp targets (msx etc), zx and ace. 

To get inverse characters you’ll need to ioctl to raw mode. 

There are a few rough spots (the function name is awful) but some feedback would be useful.

Fabrizio-Caruso, 2018-05-05:
@suborb Thanks!
The EF9345 is an odd chip but has a historical value at least in France (it was used in Minitel).
It was also used in TRS-80 MC-10 and its French clone the Aluce Matra 4k.
It is cool to have it supported.

suborb, 2018-05-07:
Information transferred to #732

closed 2018-05-07 by suborb