z88dk/z88dk #732: [udg][feature] Standard API to have udg on all targets that can do udg (including those with bitmap)

Fabrizio-Caruso, 2018-05-04:
It would be nice to have some standard way to enable udg on all targets that allowed udg
including those that can simulated them (i.e., those with bitmap mode, e.g., ZX Spectrum, CPC, etc.)

suborb, 2018-05-05:
This is being done through the "generic_console" [generic console](https://github.com/z88dk/z88dk/wiki/Classic-GenericConsole) which will interpret codes > 128 as being UDGs. 

How many UDGs are available depends on the platform (ace=32, ZX, MSX (et al) = 128)

The actual udgs are set using the IOCTL_GENCON_SET_UDGS ioctl

Fabrizio-Caruso, 2018-05-05:
@suborb thanks! 
Has the previous behavior changed? Can I preserve compatibility? Which options should I use to get the generic console vs native/previous codes?

suborb, 2018-05-05:
The branch cpc_generic_console has the start of generic console for the CPC.

Like anything involving the CPC, it's proving to be a pain, but so far there's support for all 3 modes including colour and redefinable fonts.

Fabrizio-Caruso, 2018-05-06:
@suborb great news!
I had managed to do UDG with cpcrslib both as an external lib as well as an internal lib (added by @zx70 not too long ago). Have you used cpcrslib to implement the redefinable fonts?
cpcrslib also supports software (color) sprites. It would be cool to have a wrapper to use Z88dk standard monochromatic sprite routines.

suborb, 2018-05-06:
It’s custom written so doesn’t depend on cpcrslib - it bypasses the firmware apart from the mode change code. Font can be changed with CRT_FONT like the other targets. 

I still need to do the scroll and vpeek code. The latter of which I’m not looking forward to.

zx70, 2018-05-06:
I'm very happy that my initial work on cpcrslib is becoming useful. 
there's a functionality I left behind which I'd like to introduce correctly,  and I'll need your help.. I'll try to recap it on yet another issue :S

suborb, 2018-05-07:
CPC code was merged as #737 - supports all 3 modes and vpeek is working in them as well.

suborb, 2018-05-08:
It's pretty low res on the VZ200 though - 128 x 64 pixels so just 16 x 8 standard characters, so half the playing area of the text mode.

The MC-1000 with the same chip has a few more options: https://sites.google.com/site/ccemc1000/video/modos-de-video (I've implemented the first and last modes)

zx70, 2018-05-09:
> It's pretty low res on the VZ200 though
I have your same opinion, I put the EG2000 in the list but the problem is similar.

The MC-1000 in high resolution was a little odd because the color TV output was displaying its own color  interpretation of the high resolution patterns, but more than acceptable:
https://www.youtube.com/watch?v=wAuc95OCtQ8

Fabrizio-Caruso, 2018-05-09:
@zx70 the VZ200 could display higher resolutions if equipped with more videoram

Fabrizio-Caruso, 2018-05-09:
@suborb It seems clib=ansi is broken everywhere. I think almost all Z88DK targets no longer compile with this option.
I have opened a specific issue.

@suborb it would help a lot having my project as a compile no-reg test. I am covering lots of cases and compile options.

suborb, 2018-05-11:
> I have your same opinion, I put the EG2000 in the list but the problem is similar.

So, the EG2000 is meant to support 128 user definable characters. Thus it should be possible to do the same trick as on the Sorcerer: use 32 for UDGs and 96 for a custom font, or 128 custom udgs.

However, it looks like the Genieous emulator only supports mapping 64 of them which is a little annoying - I've left the font code in there in case there's a another emulator it can be verified on.

This will be at the full 40x24 (though it could be 25) resolution.

zx70, 2018-05-12:
We're now used to see similar things but they *are* special, it would be an interesting goal even for the added vt52 capabilitiy improvements on that single target, but there's so much more!
I wonder if having a way to get the exact look and feel (including font and colours)  will be understood as a handy option or as a z88dk limit,  as it happened with the interrupts on the cpc..  :)

suborb, 2018-05-12:
For things like a console, it's pretty much a no-brainer to have a consistent standard across platforms. I think we're now at the place where we could probably create a console version of dstar which uses redefined graphics when available.

Given that it's extensible on a per platform basis (see the ZX version), if someone wants to, they can add support for multicolour characters on the CPC/MSX etc

suborb, 2018-05-12:
I've got a bit more work to do, but taken the plunge on the vg5k.

Here's ansivt52 with the CPC font:

[image]

Fabrizio-Caruso, 2018-05-12:
UDG/Tile graphics  is less than bitmap graphics and covers more platforms. Even pseudo-sprites could be implemented through UDG. Simple characters is even less. That is why my generic apis are based on characters/tiles and I can cover any 8 bit system with any CPU and graphics capabilities by rolling back to udg or ascii/petscii/atascii whenever no udg are possible.

Besides, I must recall all of you that we could have or at least simplify cross toolkit portability by providing a conio interface on top of whatever is implemented at the console level.
I personally do not like conio itself but it is used by CC65, by WinCMOC and there is an implementation for GCC for TI (TI 99/4A). Conio should just be a set of macros and have no extra computational cost and also no real implementation.

suborb, 2018-05-12:
Yes, conio applies across the vt100 and the vt52 terminal. The only thing that isn't portable is the vtrendition() function.

I may be able to do something with it to make it work with vt52, but I'm not sure at the moment - #692 makes things not so simple

Fabrizio-Caruso, 2018-08-15:
@suborb SVI has wrong WHITE color  (shown as YELLOW)
VG5K has correct colors now

Fabrizio-Caruso, 2018-08-28:
@suborb, the Tatung Einstein should also be in the list. Shouldn'it?

Fabrizio-Caruso, 2018-08-31:
@suborb I am trying to understand how to use udg with the Spectrum target.
I am getting wrong font size, no udgs and wrong colors.
Which pragmas do I need for the Spectrum?

Fabrizio-Caruso, 2018-09-09:
The Microbee can do UDG in text mode and also had high resolution bitmapped modes.

suborb, 2019-02-09:
I'm going to close this one, since this api is added for all new machines where possible.

closed 2019-02-09 by suborb

zx70, 2021-06-11:
A trick I'm using on the ZX81 to set UDGs (and elsewhere):

```
#ifdef __CONIO_VT100
int font_position() {
#asm
	extern ansifont
	ld hl,ansifont
#endasm
}
#endif

#ifdef USE_UDGS
  void *param = &udgs;

#ifdef __CONIO_VT100
  //extern unsigned char font_8x8_bbc_system[];
  //memcpy(font_8x8_bbc_system + 1536, param, 56);
  
#ifdef __ZX81__
  memcpy(font_position() + 768, param, 56);
#else
  memcpy(font_position() + 1536, param, 56);
#endif

#else
  console_ioctl(IOCTL_GENCON_SET_UDGS, &param);
#endif

#endif

```

Fabrizio-Caruso, 2023-10-14:
@zx70 is the trick above usable somehow on the Commodore 128 80col mode?

zx70, 2023-10-16:
I don't think so.
A POKE from the C128 BASIC (thus, done with the 6510 CPU)  will switch to an OUT instruction on the Z80 side.

By the way we had quite a bit of progress on that, IIRC:
https://z88dk.org/forum/viewtopic.php?p=19029#p19029