z88dk/z88dk #734: [optimization][question/feature] Can the optimizer interpret local variables as static ones as an option?

Fabrizio-Caruso, 2018-05-04:
Is there an option to convert local variables into static ones in order to gain speed and memory (at the cost of re-entrancy and therefore some compatibility).
CC65 has an option that does that and you get significant speed-ups and smaller size.
It is clearly an option that breaks ANSI C but it is something one chooses explicitly and it can do miracles in memory-tight situations. You also get significant speed improvements.

aralbrec, 2018-05-05:
Making variables static prevents zsdcc from being able to hold values in registers which will always lead to faster and smaller code.  It depends on the code and complexity of code but I would expect that blindly assigning stuff to static variables would make the code worse.

sccz80 is not able to hold values in registers across statements so it would benefit more from making variables static.

To answer the question, there is no such option.  But for your own code you could just declare things static?

Fabrizio-Caruso, 2018-05-05:
What CC65 does is slightly different than just making all local variables as static because normal static initialization is done once and with the static local option it is performed always. This makes the code behave in most cases as if the variables where not static.

Fabrizio-Caruso, 2018-05-05:
@aralbrec If I understand well, zsdcc is already using static memory for local variables and not a stack. If that is the case, this optimization is already performed.
If I understand well, sccz80 is using a stack. Am I right? sccz80 can then get a significant speed up if static memory is used instead of a stack.
To have this optimization behave as close as possible as standard C, initialization of location variables should be performed each time the function is entered (not just like static local initialization).

Fabrizio-Caruso, 2018-05-06:
@aralbrec is zsdcc by default producing non-reentrant code (i.e., incompatible with recursions and interrupts). I guess this is not the case. So I think that this optional optimization is not available in either zsdcc nor sccz80. Am I right:

What I propose is to use the BSS section instead of the stack, as a compile option. 
This is somehow similar to declaring local variables as static but initialization would also be performed once entered the function. All the scope would be checked normally.

Any code that does not use recusion and does no use interrupts, should work just fine, gain in speed and in reduced size.
You pay high price: you are no longer ANSI C compatible because you cannot longer use recursion and interrupts safely. It is hard to have the compiler check whether this optimization is possible and that is why, it could make sense to have it as a compile option (maybe with a warning about the fact that you are abusing ANSI C and losing re-entrancy).

aralbrec, 2018-05-07:
No, both compilers adhere as closely to the C standard as possible.  The 6502 has some limitations that the z80 doesn't that makes it harder to stick to the C model efficiently.

sccz80 has locals on the stack and accesses it via stack pointer manipulation.  zsdcc has locals on the stack and accesses it via stack frame in IX using indexed addressing.

zsdcc can hold locals in a register so that it never gets placed in memory.  This will always lead to the fastest and smallest code so forcing it to use static memory can make performance worse by forcing it to load/store that local variable to a memory location while running.

For example this code:

```
void foo(unsigned char) __z88dk_fastcall;

void main(void)
{
   for (unsigned char i = 0; i != 255; ++i)
      foo(i);
}
```

Is often translated to something like this:

```
_main:
	ld	c,0x00
l_main_00103:
	ld	a, c
	inc	a
	ret	Z
	push	 bc
	ld	l, c
	call	_foo
	pop	bc
	inc	c
	jr	l_main_00103
```

Here the local variable `i` is kept in register `C` and is never assigned memory on the stack.

If I then make `i` a static local variable so that it exists in memory:

```
void foo(unsigned char) __z88dk_fastcall;

void main(void)
{
   static unsigned char i;

   for (i = 0; i != 255; ++i)
      foo(i);
}
```

The compiler is forced to place `i` in memory which leads to worse code:

```
_main:
	ld	hl,_main_i_1_3
	ld	(hl),0x00
l_main_00102:
	ld	hl,(_main_i_1_3)
	call	_foo
	ld	hl,_main_i_1_3
	ld	a,(hl)
	inc	a
	ld	(hl),a
	inc	a
	jr	NZ,l_main_00102
	ret
```

As complexity increases, zsdcc has more trouble keeping locals in registers like this.  If code is written carefully by keeping functions small, using sub-function scopes, etc, the best code comes out when a mix of static locals and automatic locals is used.  You're not automatically getting better code by using static locals.

I can see maybe better code can come out if the function parameters are sent via static memory.  zsdcc will not put those values in a register unless it is told to by an assignment statement in the function, so it will access those parameters via slower (ix+n) addressing.  It will try to limit the amount of accesses it needs to do this way but this happens with mixed success.  We do have fastcall linkage which both sccz80 and zsdcc can generate code for where one function parameter is passed via register.  Then zsdcc will try to keep it in a register and will succeed if the function is not too complicated.  Then even the function call linkage doesn't use the stack.  But this is for a restricted class of functions.

I can see maybe a function decoration to specify function params are passed via static memory and another that makes all locals static with the right initialization semantics maybe could be better, the latter one more use with sccz80 I think.

Fabrizio-Caruso, 2018-05-07:
@aralbrec  CC65 is ANSI compliant with normal optimization options. 
Its "static" optimization is an extra option to get faster and shorter code at the cost of reentrancy.
It is an option to prevent the use of the stack. It is safe when reentrancy is not required.
It breaks everything in other cases. It is never used automatically.

Fabrizio-Caruso, 2018-05-07:
@aralbrec I am not an Assembly expert in either 6502 nor Z80. I know 6502 could use page zero as static local memory. It has plenty of room for many variables. 
It is up to you to decide whether having a non-reentrant option for Z88DK makes sense. It does make a lot of sense in CC65.

closed 2018-05-07 by Fabrizio-Caruso

aralbrec, 2018-05-09:
There may be savings in passing function parameters via static variables for both compilers.

For sccz80 making locals static might have advantages too but for zsdcc blindly applying this will often lead to worse code.

I think it might be worthwhile to consider two new function decorations, one for static parameter function passing and another for locals made static.

Fabrizio-Caruso, 2018-05-09:
@aralbrec  thanks! The optimization I was referring to was about avoiding allocating lcal variables (not parameters) on the stack. Not using the stack for local variables may imply breaking ANSI compliancy in some cases (e.g., recursion). It does little miracles in CC65 in terms of speed and size but it may be a 6502-related optimization.