z88dk/z88dk #736: [SVI & others] sleep not linked but prototype found

Fabrizio-Caruso, 2018-05-06:
I do not understand how I can use the sleep function with Z88DK,
I see it is included stdlib.h. I do include stdlib.h which has the prototype.
The compiler complains about not finding sleep...
If I provide my own implementatin it complains that it is already defined...

```
/cygdrive/c/z88dk/bin/zcc.exe +svi \
-SO3 --max-allocs-per-node200000 \
-compiler=sdcc \
-zorg=49152 \
-clib=ansi \
-pragma-define:ansicolumns=32 \
-pragma-include:src/../cfg/zpragma_clib.inc \
-vn -lndos \
-D__SVI__ \
-DSOUNDS \
-create-app -o build/LIGHT_svi_318 \
src/psg/psg_sounds.c    \
src/item.c \
src/display_macros.c \
src/enemy.c src/invincible_enemy.c src/level.c src/character.c \
src/text.c src/missile.c src/strategy.c src/input_macros.c \
src/main.c
src/character.c:214: warning 84: 'auto' variable 'safe' may be used before initialization
src/main.c:718: warning 126: unreachable code
Error at file 'src/character.c' line 725: symbol '_sleep' not defined
Error at file 'src/main.c' line 893: symbol '_sleep' not defined
Error at file 'src/main.c' line 1127: symbol '_sleep' not defined
Error at file 'src/main.c' line 1130: symbol '_sleep' not defined
Error at file 'src/main.c' line 1177: symbol '_sleep' not defined
Error at file 'src/main.c' line 1214: symbol '_sleep' not defined
Error at file 'src/main.c' line 1218: symbol '_sleep' not defined
Error at file 'src/main.c' line 1232: symbol '_sleep' not defined
8 errors occurred during assembly
Errors in source file c:\Z88DK\lib\config\..\..\\lib\target\svi\classic\svi_crt0.asm:
Error at file 'src/character.c' line 725: symbol '_sleep' not defined
                   ^ ---- (null)Error at file 'src/main.c' line 893: symbol '_sleep' not defined
                   ^ ---- (null)Error at file 'src/main.c' line 1127: symbol '_sleep' not defined
                   ^ ---- (null)Error at file 'src/main.c' line 1130: symbol '_sleep' not defined
                   ^ ---- (null)Error at file 'src/main.c' line 1177: symbol '_sleep' not defined
                   ^ ---- (null)Error at file 'src/main.c' line 1214: symbol '_sleep' not defined
                   ^ ---- (null)Error at file 'src/main.c' line 1218: symbol '_sleep' not defined
                   ^ ---- (null)Error at file 'src/main.c' line 1232: symbol '_sleep' not defined
                   ^ ---- (null)make: *** [Makefile:767: svi_318] Error 1
```

Fabrizio-Caruso, 2018-05-06:
If I try to implement my own sleep, using void sleep(int) as signature, I get:
```
...
src/sleep_macros.h:42: error 91: extern definition for 'sleep' mismatches with declaration.
c:/Z88DK/lib/config/../..//include/stdlib.h:295: error 177: previously defined here
make: *** [Makefile:767: svi_318] Error 1
```

It looks like as if the sleep name were taken but it cannot be used for anything.

suborb, 2018-05-06:
I had a quick look, it appears I forget to export the sdcc symbol when I jiggled things around. I've added so it'll be sorted in the next nightly.

commit d9fa868a (2018-05-06): Fix #736: Export _sleep for sdcc
closed 2018-05-06 by suborb