z88dk/z88dk #738: z80asm: superfluous section generated in compile

aralbrec, 2018-05-08:
I just committed an [example](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/zxn/sprites/mixed-sw-hw) where the compile generates an extra file "main" with two bytes in it.  Usually this means something in the project was not assigned to a section, however I can't find it.  The extra file causes appmake to error without building the output.

Compile line:
```
zcc +zxn -v -startup=1 -clib=sdcc_iy -SO3 --max-allocs-per-node200000 main.c graphics.asm.m4 interrupt.asm -o main -pragma-include:zpragma.inc -subtype=sna -Cz"--clean" -create-app -no-cleanup --list -m
```

An extra file "main" is generated that contains 2 bytes.

The error:
```
zxn: Error: Section  has org in rom 0x0000
Building application code failed
```

Appmake picks up `main` and assumes it belongs to the unnamed section "" which generates this error.

I can generate the output file `main.sna` if I delete `main`, copy the map file out of the temp directory and rename it `main.map`, and then:

```
appmake +zxn --sna -b main -o main.sna --clean
``

This requires a zx next emulator to run and it seems to run properly without the extra 2 bytes.

I tried a few things to locate where these 2 bytes are coming from.

1.  I compiled a consolidate object file by adding `-c` to the compile line.  A z80nm on the object file showed no extra sections with two bytes in it.  The consolidated object file contains everything except the crt.

2. The crt being used is [zxn_crt_1](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_1.asm.m4).  This crt includes files with defines in them before the memory map is defined and before the first section is opened.  I added a `SECTION BLAH; org 0` to the top of the crt so that anything that might not be in a section would get added to output binary `main_BLAH.bin` instead.  **Another compile produced two extra identical binary files containing 2 bytes "main" and "main_BLAH.bin".  This doesn't make sense so it appears to be a bug in z80asm.**

3. I tried looking at the crt's .lis file in the temp directory.  No bytes are generated outside the defined sections.

4. I tried look at the map file and the section sizes.   A few sections contained two bytes but they are part of the memory map.

Because of 2 above, it seems to be a bug in z80asm.

pauloscustodio, 2018-05-09:
Thanks for reporting. I've been away this week but will look at it when I come back.

pauloscustodio, 2018-05-11:
I tried to test in Ubuntu but I don't have zsdcc and zsdcpp. In windows I usually grab these from the nightly build.

How can I build these in Ubunty?

aralbrec, 2018-05-12:
Oh yeah, currently there is #730 which says the patch needs to be applied in a sub dir.

aralbrec, 2018-05-12:
I still get the error:

```
appmake +zxn --sna  -b "main" -c "...\Local\Temp\zcc45D25"
zxn: Error: Section  has org in rom 0x0000
Building application code failed
```

The binaries I am using are built with VS2015.  If you want to use mine:  [z88dk/bin](https://drive.google.com/file/d/19eI7_7HmgXUsR0aM2M7QG39Fbj1CFuep/view?usp=sharing)

I do have a number of things not committed but I don't think it affects this.

aralbrec, 2018-05-12:
Maybe we can ask @suborb or @zx70 to try to see if they have any problems?

suborb, 2018-05-12:
This compiles to .sna without any problems for me (Mac)

aralbrec, 2018-05-12:
I'll have a closer look at what I haven't committed yet.

aralbrec, 2018-05-15:
I can't find anything that would cause these extra two bytes to be generated into the unnamed section.

@pauloscustodio Is there a good place in the z80asm source to identify when a few bytes are placed into the unnamed section?

aralbrec, 2018-05-16:
I'm not getting anything placed in the unnamed section.

I modified appen_value() like this:
```
void append_value( int value, int num_bytes )
{
    init_module();

    if (strlen(g_cur_section->name) < 5)
    {
        SymbolHash *symbols;
        SymbolHashElem *iter;
        Symbol *lastsym;

        fprintf(stderr, "\n\nSection \"%s\", %d bytes\n", g_cur_section->name, num_bytes);

        symbols = select_module_symbols(get_cur_module(), blahblah);

        lastsym = 0;
        for (iter = SymbolHash_first(symbols); iter; iter = SymbolHash_next(iter))
            lastsym = (Symbol *)iter->value;

        if (lastsym)
            fprintf(stderr, "Last symbol = %s\n", lastsym->name);

        fprintf(stderr, "\n");
    }

	patch_value(get_cur_module_size(), value, num_bytes);

	if ( opts.list )
		list_append( value, num_bytes );
}
```

Output is this:

```
Section "CODE", 1 bytes
Last symbol = __Start

Section "CODE", 1 bytes
Last symbol = __Start

Section "CODE", 1 bytes
Last symbol = __Start

Section "CODE", 1 bytes
Last symbol = __Start
```

And I can see those four bytes added to section CODE in the crt.lis.

I'll try to find where z80asm generates output next.

aralbrec, 2018-05-16:
Modifying fwrite_codearea() in codearea.c to see the sections as they are output:
```
		/* bytes from this section */
		if (section_size > 0 || section->origin >= 0 || section->section_split)
		{
            if (section_size > 0)
            {
                fprintf(stderr, "\n\nsection \"%s\", org %d, size %d\n\n", section->name, section->origin, section_size);
            }
```

The two bytes in the unnamed section are there:

```
section "", org -1, size 2
section "CODE", org 30000, size 4
section "code_crt_init", org -1, size 42
```

However a "-Cl-v" for verbose output from z80asm does not list the unnamed section.

aralbrec, 2018-05-16:
Is there another place besides append_value() where these two bytes could be added?

pauloscustodio, 2018-05-16:
> Is there another place besides append_value() where these two bytes could be added?

Not in normal conditions, but there might be a bug somewhere.

Can you built a zip of your whole tree and instruction how to generate the error? I can try to look into z80asm from your tree.

aralbrec, 2018-05-16:
Current z88dk tree minus .git:
https://drive.google.com/file/d/1KokuxlnnFr8SaPPiOJzSwwauuB6exrWy/view?usp=sharing

I also removed the changes made to print diagnostic information as detailed in the last few messages.

I think the error is occurring for all compiles or nearly all compiles but for sure you can build the example in [z88dk/libsrc/_DEVELOPMENT/EXAMPLES/zxn/sprites/mixed-sw-hw](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/zxn/sprites/mixed-sw-hw)

```
zcc +zxn -v -startup=1 -clib=sdcc_iy -SO3 --max-allocs-per-node200000 main.c graphics.asm.m4 interrupt.asm -o main -pragma-include:zpragma.inc -subtype=sna -create-app
```

You will get this error:

```
zxn: Error: Section  has org in rom 0x0000
Building application code failed
```

because appmake is finding the file "main" which is where the two bytes from the unnamed section are written.  This file should be deleted between compiles because its presence will always generate this error whether the current compile created it or not.

Faster zsdcc compile:
```
zcc +zxn -v -startup=1 -clib=sdcc_iy -SO3 main.c graphics.asm.m4 interrupt.asm -o main -pragma-include:zpragma.inc -subtype=sna -create-app
```

sccz80 compile:
```
zcc +zxn -v -startup=1 -clib=new main.c graphics.asm.m4 interrupt.asm -o main -pragma-include:zpragma.inc -subtype=sna -create-app
```

You can add `-m --list` to get the map file out and the list files, including the crt in `main.lis`

pauloscustodio, 2018-05-17:
> rWy/view?usp=sharing

Thanks, I managed to reproduce the error.

The error is caused by the lines at config_zxn_public.inc lines 4053 to
4062, where symbols are defined as themselves. This causes an assertion
failure during linking, which is not caught in release mode.

Is this a typo?

For fixing the error I can add an error message in DEFC when it detects
that the to-be-defined name is used in the expression.

aralbrec, 2018-05-17:
> The error is caused by the lines at config_zxn_public.inc lines 4053 to 4062, where symbols are defined as themselves. This causes an assertion failure during linking, which is not caught in release mode. Is this a typo?

Indeed it is.  I'll fix it up and verify it's working.

aralbrec, 2018-05-17:
All good!

commit dcd28421 (2018-05-17): Fix #738: Some defines missing from zxn configuration
closed 2018-05-17 by aralbrec