z88dk/z88dk #740: [spectrum] Previous code with clib=ansi no longer compiles

Fabrizio-Caruso, 2018-05-09:
Previous code with clib=ansi no longer compiles.
I may come from the generic console.

```
$ make spectrum_48k
/cygdrive/c/z88dk/bin/zcc.exe +zx -O3 -clib=ansi -vn  \
-pragma-redirect:ansifont=_udg -pragma-define:ansifont_is_packed=0 -pragma-define:ansicolumns=32 \
-DFULL_GAME -DREDEFINED_CHARS -DSOUNDS -DCLIB_ANSI -DEND_SCREEN -DBETWEEN_LEVEL -D__SPECTRUM__ \
-lndos -create-app \
-o build/FULL_spectrum_48k.prg \
src/horizontal_missile.c src/rocket.c src/item.c src/end_screen.c \
src/spectrum/udg.asm src/spectrum/spectrum_graphics.c src/display_macros.c \
src/enemy.c src/invincible_enemy.c src/level.c src/character.c \
src/text.c src/missile.c src/strategy.c src/input_macros.c \
src/main.c
Error at file 'src/end_screen.c' line 135: symbol 'vtrendition' not defined
Error at file 'src/spectrum/spectrum_graphics.c' line 429: symbol 'vtrendition' not defined
Error at file 'src/spectrum/spectrum_graphics.c' line 502: symbol 'vtrendition' not defined
Error at file 'src/spectrum/spectrum_graphics.c' line 554: symbol 'vtrendition' not defined
Error at file 'src/spectrum/spectrum_graphics.c' line 617: symbol 'vtrendition' not defined
Error at file 'src/text.c' line 93: symbol 'vtrendition' not defined
Error at file 'src/text.c' line 136: symbol 'vtrendition' not defined
Error at file 'src/text.c' line 163: symbol 'vtrendition' not defined
Error at file 'src/text.c' line 191: symbol 'vtrendition' not defined
Error at file 'src/text.c' line 244: symbol 'vtrendition' not defined
Error at file 'src/text.c' line 271: symbol 'vtrendition' not defined
Error at file 'src/text.c' line 298: symbol 'vtrendition' not defined
Error at file 'src/text.c' line 325: symbol 'vtrendition' not defined
Error at file 'src/text.c' line 366: symbol 'vtrendition' not defined
Error at file 'src/text.c' line 392: symbol 'vtrendition' not defined
Error at file 'src/text.c' line 564: symbol 'vtrendition' not defined
Error at file 'src/text.c' line 620: symbol 'vtrendition' not defined
Error at file 'src/text.c' line 641: symbol 'vtrendition' not defined
Error at file 'src/text.c' line 659: symbol 'vtrendition' not defined
Error at file 'src/main.c' line 305: symbol 'vtrendition' not defined
Error at file 'src/main.c' line 729: symbol 'vtrendition' not defined
Error at file 'src/main.c' line 776: symbol 'vtrendition' not defined
22 errors occurred during assembly
Errors in source file c:/z88dk/\lib\target\zx\classic\spec_crt0.asm:
Error at file 'src/end_screen.c' line 135: symbol 'vtrendition' not defined
                   ^ ---- (null)Error at file 'src/spectrum/spectrum_graphics.c' line 429: symbol 'vtrendition' not defined
                   ^ ---- (null)Error at file 'src/spectrum/spectrum_graphics.c' line 502: symbol 'vtrendition' not defined
                   ^ ---- (null)Error at file 'src/spectrum/spectrum_graphics.c' line 554: symbol 'vtrendition' not defined
                   ^ ---- (null)Error at file 'src/spectrum/spectrum_graphics.c' line 617: symbol 'vtrendition' not defined
                   ^ ---- (null)Error at file 'src/text.c' line 93: symbol 'vtrendition' not defined
                   ^ ----       #define GUN_IMAGE_Y 0
Error at file 'src/text.c' line 136: symbol 'vtrendition' not defined
                   ^ ----
Error at file 'src/text.c' line 163: symbol 'vtrendition' not defined
                   ^ ----
Error at file 'src/text.c' line 191: symbol 'vtrendition' not defined
                   ^ ----                       SET_TEXT_COLOR(TEXT_COLOR);
Error at file 'src/text.c' line 244: symbol 'vtrendition' not defined
                   ^ ----                       PRINTF(LEVEL_X,1-Y_OFFSET,"%02u", level);
Error at file 'src/text.c' line 271: symbol 'vtrendition' not defined
                   ^ ----               #else
Error at file 'src/text.c' line 298: symbol 'vtrendition' not defined
                   ^ ----
Error at file 'src/text.c' line 325: symbol 'vtrendition' not defined
                   ^ ----               _printScoreOnRow((YSize>>1), text, score);
Error at file 'src/text.c' line 366: symbol 'vtrendition' not defined
                   ^ ----
Error at file 'src/text.c' line 392: symbol 'vtrendition' not defined
                   ^ ----               printCenteredMessageOnRowWithCol(3, _RED,  CROSS_CHASE_STRING);
Error at file 'src/text.c' line 564: symbol 'vtrendition' not defined
                   ^ ---- (null)Error at file 'src/text.c' line 620: symbol 'vtrendition' not defined
                   ^ ---- (null)Error at file 'src/text.c' line 641: symbol 'vtrendition' not defined
                   ^ ---- (null)Error at file 'src/text.c' line 659: symbol 'vtrendition' not defined
                   ^ ---- (null)Error at file 'src/main.c' line 305: symbol 'vtrendition' not defined
                   ^ ----
Error at file 'src/main.c' line 729: symbol 'vtrendition' not defined
                   ^ ---- (null)Error at file 'src/main.c' line 776: symbol 'vtrendition' not defined
                   ^ ---- (null)make: *** [Makefile:1085: spectrum_48k] Error 1
```

suborb, 2018-05-09:
I think vtrendition has gone missing from the libraries - I think when I was sharing code I accidentally removed it.

Sorry, it'll be back by tomorrow.

suborb, 2018-05-09:
It's now been re-added, so I'll close this one.

closed 2018-05-09 by suborb