z88dk/z88dk #741: [no-regs][compile][feature] Compile No-reg tests

Fabrizio-Caruso, 2018-05-09:
It would be cool to have some compile no-reg tests that are run after each PR to test whether all or most compile options on all targets work.

This could be done with some trivial code (e.g., "hello world"), just to test compilation.

I could help on this by starting with my large Makefile and using just some "hello world" code instead of my game's code.

Integration of my game (CROSS CHASE) or a simplified version of it would also be cool but it could make the compilation no-reg tests too slow.

suborb, 2018-05-09:
Pretty much all of the targets have their examples built as part of the build.

To increase coverage, the build also compiles example/console/world.c and example/console/ansitest.c for as many targets as possible.

Fabrizio-Caruso, 2018-05-09:
Maybe we need more than just a hello world.
How was the vtrendition problem missed by the tests?

Fabrizio-Caruso, 2018-05-09:
My offer about a slim version of CROSS CHASE for Z88DK is still valid.
I can make it small and clean and I won't update it every other ten seconds as I do with my project.
I could just maintain it by fixing bugs if and when they are found.

closed 2018-05-09 by Fabrizio-Caruso

suborb, 2018-05-09:
@Fabrizio-Caruso I'd love it if you could prepare a cut down version - your project is great test case due to the number of platforms and depth of functionality.

vtrendition() is a shortcut that Stefano added to quickly change the attributes - ansitest uses the embedded sequences.

Fabrizio-Caruso, 2018-05-09:
@suborb What the max raw binary size for the average target I should consider?
The raw binary takes 
- about 14k for the full version (no UDG and maybe no sound)
- about 8k for a lighter version of the game

suborb, 2018-05-09:
I don't think the size matters so much, probably best to go with whatever config you've found that works to be honest.

It's probably best just to go for a "release branch" version of your repo - that way it's minimal extra maintenance effort for you.

Fabrizio-Caruso, 2018-05-13:
@suborb  @zx70  I don't want to bloat Z88DK with the current full code of my game.
I will have to clean it up and probably make a Z80-only version the can be easily upgraded it to the current universal one currently supporting 6502/6809/Z80 (more are scheduled).
I may also remove the graphics data to make it small and provide the instructions on how to upgrade it to the full version.

suborb, 2018-05-13:
The idea I had regarding this was to just build nightly - the nightly machine would pull down a branch directly from your repo. 

This way it’s not a burden on you - apart from having a release branch where you release from.

Fabrizio-Caruso, 2018-05-14:
@suborb  OK! Once I get a clean version I will ceeate a release branch. I guess you will pull all of it. I could simply tell you which directories contain 6502 and 6809 related code. You could simply exclude those to reduce Z88DK's size increase .

Fabrizio-Caruso, 2018-05-14:
@suborb I have just found out that git supports sparce checkouts: you can selectively check out the Z80 directories of my project and leave out the others.