z88dk/z88dk #748: [question] When should I use -lndos and when not?

Fabrizio-Caruso, 2018-05-13:
If my program does not read from any disk/tape should I always use -lndos ?
Even when compiling with sdcc?

aralbrec, 2018-05-14:
It's a classic library requirement so it doesn't make a difference which compiler you are using.

I'm not sure if @suborb has changed this in the classic library with everything that is going on but it used to be that the classic's stdio implementation defined stdin ,stdout, stderr ,string, network and disk streams.  If you used stdio functions (like printf) the target had to supply network and disk drivers as well.  If the program doesn't use them or the target doesn't support them, "-lndos" is added to the compile line to provide empty stubs for network and disk.

On the other hand, many systems had several incompatible disk hardware.  The spectrum, eg, had dozens of different disk systems none of where were compatible.  For the spectrum then, you would decide which one you were compiling for replacing "-lndos" with something like "-lp3" which would select one particular disk system, in this case the spectrum +3's built in disk drive.

suborb, 2018-05-14:
Yes, -lndos is still needed - given the changes in zcc/z80asm lately I could drop the requirement for it fairly easily.

However, I've got a few ideas around fileio that I'd like to do at the same time, unfortunately #692 clobbers them.

You can bypass the need to include any "traditional stdio" by compiling with -pragma-define:CRT_ENABLE_STDIO=0 and using printk instead of printf

closed 2018-05-15 by suborb