z88dk/z88dk #750: (classic) amalloc not set up correctly for most targets

suborb, 2018-05-15:
The amalloc initialisation code needs hl to contain the highest memory address that the heap can extend to before it's called.

Unfortunately, most targets don't set this correctly.

zx70, 2018-05-15:
Thankyou, I wasn't fully confident but I suspected it..
  https://github.com/z88dk/z88dk/issues/498#issuecomment-381204090

suborb, 2018-05-15:
Yeah, I spent quite a while trying to debug random failures in ScottFree until I figured out it was the heap trashing the code.

I've fixed up a few of the targets that I know have different stack setups, for the rest of them I'm assuming that sp > BSS_END and we own the memory - I'm sure it's the wrong assumption, but we can fix them easily enough. The work around without changing the CRT is to add the option `-pragma-define:CRT_MAX_HEAP_ADDRESS=nnnnn`

zx70, 2018-05-17:
ScottFree ..  :)
...years ago I got it running on real Microdrives (fcntl interfacing directly at sector level), loading speed wasn't worse than on the emulators and the files (which I uploaded from rs232 with simple BASIC commands) were perfectly read.   No need to say that also the ANSI VT emulation worked well .

Are you adapting it to the universal console ?

suborb, 2018-05-17:
I was contemplating porting a curses game to the new console and then I got sidetracked by ScottFree again.

Ah, that's the example using a variable for the heap isn't it? My fix is just when you use -DAMALLOC 

I'll take a look at the variable variant code.

zx70, 2018-05-18:
The problem I have is in {z88dk}/libsrc/lib3d/mesh_new.c
I temporairly removed the malloc() calls in the MSX examples and I'm using static buffers everywhere else.

zx70, 2019-10-01:
I think the AMALLOC is now in a much better shape, but I can't say if it is fixed on all the targets.
Thoughts ?

closed 2020-02-02 by suborb