z88dk/z88dk #751: [vz200] BASIC cursor sometimes remains active and so prevents input to work

Fabrizio-Caruso, 2018-05-17:
Sometimes I am getting a random weird error with VZEM that I can fix by reloading my game: the cursor appears and so prevents the game input to work by remaining active during the game. 
What is going wrong? Is it an emulator issue?

I tend to get this error with the 18k configuration in VZEM.

Remark: I am using getk to read the input.

Fabrizio-Caruso, 2018-05-18:
It could be an emulator issue. It seems other people have observed this.

suborb, 2018-05-23:
I think it's going to best to bypass the firmware for reading keyboard and implement input/in_Inkey etc, there's an appropriate chapter here: http://messui.polygonal-moogle.com/comp/vz200/vz200_trm.pdf

>As explained earlier, the 45 keys of the VZ-200 keyboard are arranged in an 8 x 6 matrix scanned by the eight least significant address lines. This means that each row of the keyboard matrix effectively occupies a specific memory location (or more accurately, a series of memory locations, due to the simplified decoding). The individual keys are mapped into the corresponding bits of that memory location, according to the column they occupy in the matrix.

> The keyboard is scanned by the software taking each of the eight address lines down to logic low level in turn. If a key is pressed, it effectively "pulls down" the bit line at the address for its row. For example if the "2" key is pressed, this causes bit 1 to drop to 0 at address 68F7 (also 69F7, 6AF7, 6BF7, 6CF7, 6DF7, 6EF7 and 6FF7). Providing no other keys are pressed in that row, the data retrieved by reading that address will be 3D hex (binary 111101).

> The keyboard matrix and its (lowest) row addresses in hex are shown below. Note that each key causes a logic 0 to appear at the bit position shown, when its row address is read.

[image]

Fabrizio-Caruso, 2018-05-24:
Yep, @suborb, I was going to implement exactly that matrix in my code unless a solution is included in Z88DK.
Using kernel routines is a smart way to save memory size but I understand they can be a pain in the neck to use due to their interactions with the rest of the system.

Fabrizio-Caruso, 2018-05-24:
@suborb, what is puzzling is that the current solution works at random times. Usually reloading the same binary fixes the issue. I don't know how reliable VZEM emulator is, though.

suborb, 2018-05-24:
I've stuck this in queue behind sorting out SC-3000/SG-1000 support.

suborb, 2018-05-25:
I've added in_Inkey() support for the vz - hopefully that'll solve your problems.

Let me know!

closed 2018-05-25 by suborb

Fabrizio-Caruso, 2018-05-26:
@suborb the cursor seems gone!