z88dk/z88dk #752: [question] Constant division

Fabrizio-Caruso, 2018-05-18:
Can this be of any help?
https://github.com/stevenhoving/integer_division

suborb, 2018-05-23:
The code isn't particularly relevant, but the linked post: http://homepage.divms.uiowa.edu/~jones/bcd/divide.html is, could possibly pull out a few of the methods to speed some things up.

Given that it's shifts, then sanity checking against http://www.chilliant.com/z80shift.html is probably a wise idea too.

Fabrizio-Caruso, 2018-05-23:
@suborb, we should also understand when it worthwhile to sort of inline division and multiplication by a constant without paying a price in terms of size.
On targets with little ram, size is more important than speed.
I guess small powers of 2 should be handled at compilation time.

suborb, 2018-05-23:
I agree, it's the balance I've been attempting to strike in sccz80 - between speed/size. Hopefully it's worked out alright.

I wrote this a while back: https://github.com/z88dk/z88dk/wiki/WritingOptimalCode I should probably revisit and revise soonish.

Fabrizio-Caruso, 2018-05-31:
I am closing this non-issue. It is just a hint.

closed 2018-05-31 by Fabrizio-Caruso