z88dk/z88dk #757: (Classic) Sega SG-1000 support

suborb, 2018-05-20:
* Z80 at 3.579
* TMS9918 with 16k VRAM (ports 0xbe, 0xbf)
* SN76489 soundchip
* 1k RAM at 0xc000 (Cartridges can supply more)
* 2 Joysticks ports 0xDC and 0xDD (signals mixed)

Info: http://www43.tok2.com/home/cmpslv/Sg1000/EnrSG.htm

Should be fairly trivial to add support for

zx70, 2018-05-21:
It is probably not in working order, but "startup=2" in the SC-3000 was intended to do exactly what you're asking for.

suborb, 2018-05-21:
Cool. I shall tend to it when I’ve finished with the Casio. 

Probably give it a separate config file and crt0 to make it more obvious. 

It’s nice how all the VDP stuff just works across ports - we should probably do the Coleco ones to finish off the set.

Fabrizio-Caruso, 2018-05-21:
SC 3000 and SG 1000 are supposed to be compatible to a very high degree

suborb, 2018-05-23:
@zx70 what ROM are you using for the SC-3000? 

I've got "Home Basic (Japan) (SC-3000).sc" and the entry point $39e2 doesn't make a whole lot of sense in the context (changing screen mode)

I'm going to make the VDP for the SC-3000/SG-1000 output to the ports directly rather than via ROM to avoid this problem, but it seems I've not quite initialised the VDP directly (hence wanting your ROM version so that I can inspect the actual state)

suborb, 2018-05-24:
There’s no jump table in the rom I found unfortunately. 

I’ve got the gen_set_mode working now for mode 2 so I’ve bypassed it there. 

I’ll do the same for the keyboard and then we can be rom free.

suborb, 2018-05-25:
@zx70 thanks for the hint, I merged this as #768

closed 2018-05-25 by suborb