z88dk/z88dk #758: Type conversion bug in sdcc

zx70, 2018-05-21:
I had to change vtstone.c to make it run correctly when compiled with sdcc.

I had to change the expression in the original code:

int dmove(char *_new, int move)
{	int i;
	int j; 
	int who;
	if ((move < 1) || (move > 13) || (move == 7) || ! _new[move])
		printf("Bad arg to mmove: %u",move);

...to:
	if ((move < 1) || (move > 13) || (move == 7) || ! (int) _new[move])

The program was compiled with "--fsigned-char" as required.

aralbrec, 2018-05-21:
I can't reproduce a problem.  What's the compile line you are using?

```
#include <stdio.h>

extern void decpit(char *board, int pit);
extern void incpit(char *board, int pit);
extern int mod(int i,int j);
extern int sleep(int);

int dmove(char *_new, int move)
{	int i;
	int j; 
	int who;
	if ((move < 1) || (move > 13) || (move == 7) || !_new[move])
		printf("Bad arg to mmove: %u",move);
	who = (move < 7 ? 1 : 0);
	i = _new[move];
	for (j = 0; j < i; j++) decpit(_new,move);
	sleep(2);
	while (i--) {
		move = mod(move,who);
		incpit(_new,move);
		putchar(7);
		/*sleep(1);*/
	}
	if (_new[move] == 1 && who == (move < 7 ? 1 : 0) && move && move != 7)
		while(_new[14-move]) { 
			decpit(_new,14-move);
			incpit(_new,who*7);
		}
	if (move == 0 || move == 7) return(0); 
	else return(1);
}
```

`&_new[move]` is being kept in register `bc` which is computed early in the function:

```
	ld	a,(ix+4)
	add	a,(ix+6)
	ld	c, a
	ld	a,(ix+5)
	adc	a,(ix+7)
	ld	b, a
```

Then the test is this:

```
	ld	a, (bc)
	or	a, a
	jr	NZ,l_dmove_00102
```

zx70, 2018-05-22:
I can't reproduce the problem anymore !  :(
My impression is that the compilation today takes longer than yesterday, but honestly I'm not able to remind a single difference on what I did few hours ago  :(

The Makefile creates this line:

zcc +msx -vn -llib3d  -lm -lndos -create-app -compiler=sdcc  -SO3 --max-allocs-per-node200000 --fsigned-char  ../console/vtstone.c -I../console -clib=ansi -pragma-define:ansicolumns=80 -create-app -ovtstone.bin

The game started correctly but then the player's move failed with the message: "Bad arg to mmove", I'm now unable to reproduce the problem. (??)

zx70, 2018-05-22:
No luck, I checked both SVI and MSX, where I found the problem.   I also tried to leave some dirt and mix sdcc with sccz80 (e.g. trying to fool the compiler with a wrong object file)  but I can't re-create the bug.

closed 2018-05-22 by zx70

zx70, 2018-05-22:
For the record, building vtstone for the SORD M5, including the WAV file creation takes about 2.5 minutes on an I5 core, 2.6 Ghz,   but people complaining for the slowness  after some experimenting for few months gets back to me asking why we insist keeping sccz80 (??)   Interesting for comparisons:   vpoke.c