z88dk/z88dk #759: CP/M - subtypes for z180 & z80-zxn - enhancement

feilipu, 2018-05-21:
The CP/M target is specific to the z80 machine type. There is no subtype option available to support the use of z180 and z80-zxn op codes.

Whilst there may be little reason to re-assemble the libraries with alternative machine types, being able to write applications that use the machine specific enhancements is quite useful. As an example, all of the I/O register controls for the z180 are reached by the `out0` and `in0` instructions, specific to the z180.

The enhancement request is to allow both `--subtype=z180` and `--subtype=z80-zxn` options to enable their respective op codes in the `+cpm` target.

aralbrec, 2018-05-23:
I'll try to commit the changes tomorrow.

For specific targets that can run cpm like the rc2014, yaz180, zx-next I've been leaning toward adding a cpm subtype for the targets themselves.   That would allow the cpm compiled program to access all the library functions of the specific target.  Using the +cpm target restricts the library to standard cpm.

feilipu, 2018-05-23:
I was considering the lesser of two evils. Maintenance of CPM calls across multiple targets, vs changing cpu on one CPM target.

My guess was that it’s less maintenance to have one CPM target but with optional cpu subtypes, since the BDOS calls are the same. But that loses the  target specific config ability. If it is pure CPM then probably that’s the better option.

feilipu, 2018-05-24:
Ah yes. I was forgetting the C library `ixl` and related op code issues. You are right that for z180 and rabbit there will have to be an additional subtype in the target.

Currently I’m just working in assembly to BDOS calls, so the CPM z80 target solution is sufficient. But I can also think about attempting a PR for a YAZ180 CPM subtype, if you’ve not already solved it?

feilipu, 2018-05-24:
IIRC, the z80 `nop` s the illegal instructions, so a test to `xor ixl dec ixl jr C,__Start` should disqualify z180. Right?

aralbrec, 2018-05-24:
The z180 has that illegal instruction trap so I think it generates an interrupt for those.  To detect z180 I think you'd have to pick an undocumented z80 instruction that has another meaning on the z180.  Like a mlt or something and check the result.

aralbrec, 2018-05-24:
> Currently I’m just working in assembly to BDOS calls, so the CPM z80 target solution is sufficient. But I can also think about attempting a PR for a YAZ180 CPM subtype, if you’ve not already solved it?

If you'd like to have a go at it, go ahead.  The cpm target is using most of the same library code.  There might be some additional code you can find listed in the cpm target's library directory that is hopefully  namespaced with leading cpm_* which can be added to the yaz180 and other targets.  Then the drivers for cpm and the crts should be added to the yaz180's list of library routines and its crt startup directory, in a similar way the zxn target inherits code and drivers from the zx target.

aralbrec, 2018-07-11:
cheers.

I think we can use a single cpm target for the different processors by making use of -subtype on the compile line.  This will allow zcc to choose a different set of libraries depending on targetted processor.

`zcc +cpm ... -subtype=z180 ...`

But also supplying a cpm subtype for specific target machines will allow those cpm programs to use machine-specific code so that should be done as well.

`zcc +yaz180 ... -subtype=cpm ...`

aralbrec, 2018-07-11:
I'll resurrect the topic to keep it in the to-do

feilipu, 2018-07-11:
Is a full complete test for z180...

```
xor a
dec a
daa
rlca
rlca
jp NC, _z80
z180:
```
Or is there a shorter / faster way?

EtchedPixels, 2018-07-11:
Would need to check what Z280/R800 do if you care about them I guess. It depends how far you want to go 8)

For the code wouldn't it be far shorter to do

xor a
dec a
daa
jp p, ..

as according to the docs the DAA instructions sets the S flag based on the result. Without digging out my 8080 manual I can't remember if the xor a / dec a is sufficient to give a condition code check for 8080/85 v Z80 as well.

BTW reading the earlier comments the Rabbit's don't matter - they are not quite Z80 or 8080 compatible so won't run CP/M. The Z280/R800 will and do.

I don't know about Z180, but Z280 and R800 do handle the IX/IY undocumented but some of the undocumented shifts become other things. The Z280 in fact makes a lot of the unofficial ones official.

If your Z80 maths lib breaks on Z180 you also have problems on TRS80 Model 4/4P (XLR8R and various other Z180 upgrades were common)

feilipu, 2018-07-11:
Ah yes. I haven’t graduated to using the sign flag yet. Thanks.

The z180 turns undocumented into `nop`, unless the opcode is used like `mlt rr`.

We’re building `mlt rr` and `mul de` integer math, so the issue is a positive switch opportunity, rather than the slower z80 code breaking.

suborb, 2018-07-13:
From the emulator sources (mame, sdcc, ticks) I've looked at, I don't think the z180 DAA behaviour is emulated correctly.

I suspect there's a lot more broken emulators out there as well.

EtchedPixels, 2018-07-14:
Any Z80 emulator should get it right and produce 99 or F9. It's not going to pass any of the test tools otherwise. It's documented behaviour so IMHO if the "emulator" gets it wrong it's not a Z80 emulator. MAME gets it right and passes ZEXALL.

Z80 users manual page 173 in the 2016 version has the Z80 implementation table.

The easiest way think of it is to remember that internally the Z80 has a 4bit ALU so it executes the DAA in two parts. The Z180 otoh is wider. Subtracted to FF is not a valid input so neither is broken.

It's also described quite clearly in Zaks.

I'd expect jp p to work - DAA is a maths op so the sign flag should be set according to the top bit. My actual code does a CP 99 though.

If ticks gets something like DAA wrong I'd suggest you try ZEXDOC on it. ZEXDOC tests all the documented official instruction behaviour and is kind of a minimum standard for any Z80 emulator.
ZEXALL checks a lot of the undocumented stuff stuff people sometimes rely upon but is probably rather less relevant. (There's even more extreme testers that even the NEC clones fail 8))

suborb, 2018-07-14:
As expected, ticks implements DAA correctly according to zexall.com.

That's in z80 mode, in z180 mode it will still produce $99 for the test we're discussing above.

My point was that this code:

    inline void z80_device::daa()
    {
        uint8_t a = A;
        if (F & NF) {
                if ((F&HF) | ((A&0xf)>9)) a-=6;
                if ((F&CF) | (A>0x99)) a-=0x60;
        }
        else {
                if ((F&HF) | ((A&0xf)>9)) a+=6;
                if ((F&CF) | (A>0x99)) a+=0x60;
        }

        F = (F&(CF|NF)) | (A>0x99) | ((A^a)&HF) | SZP[a];
        A = a;
    }

Which is from Mame. The code is the same for both z80 and z180 cores. I can't see a path where we end up with $F9 as the exit value if the input is $FF

EtchedPixels, 2018-07-14:
Ah got you - yep I can believe some Z180 emulations are wrong, quite a few are Z80 ones with stuff pasted in or out.

feilipu, 2018-07-16:
Ok. I'm not going to pretend that I understand how two different answers to the same question can both claim to be "correct", but here's the screenshots of a real z80 `AF` register after executing `xor a` `dec a` `daa` instructions.
[image]
And for comparison, here's the z180 version.
[image]
Different, but both correct?

For interest the Z180 seems to leave the unused flag bits set to 1.

feilipu, 2018-07-17:
Had a few minutes spare, so I've done a bit more hardware testing on various combinations.
Screenshots follow. They should be self explanatory.
There are some substantial differences between the behaviour of the two CPUs.

[image]
[image]

[image]
[image]

[image]
[image]

[image]
[image]

[image]
[image]

[image]
[image]

[image]
[image]

suborb, 2018-07-17:
Thanks for running those,  a good sample of (invalid!) uses. Following the tweak committed a few days ago, ticks get the A result correct, but F still follows z80 rules.

As you mentioned it looks like F3 and F5 are always set on the z180, and the carry flag looks like it's reset for subtractions/decrements that result in "invalid" BCD.

EtchedPixels, 2018-08-30:
Just by way of info in case it matters

[http://www.zilog.com/appnotes_download.php?FromPage=DirectLink&dn=AN0066&ft=Application%20Note&f=YUhSMGNEb3ZMM2QzZHk1NmFXeHZaeTVqYjIwdlpHOWpjeTk2TVRnd0wyRndjRzV2ZEdWekwzb3hPREJ4WVM1d1pHWT0=](url)

has an entry on all the (documented) differences between Z80 and Z180 and as well as DAA there are two other small differences about RLD/RRD and flags.

feilipu, 2019-01-16:
Added a `-subtype=cpm` for RC2014 in #1066.

Will follow up with a similar PR for YAZ180.

feilipu, 2019-02-10:
Assuming that z80-zxn CP/M support will be provided in a more robust manner, now that #1066 is merged I think this can now be closed.

closed 2019-02-10 by feilipu