z88dk/z88dk #767: [question][TI85][TI86] loading them without a shell?

Fabrizio-Caruso, 2018-05-24:
If I compile for TI85 or TI86 with create-app I get both raw binaries and files that seem to be already cooked to be loaded even without a shell.
The TI86EMU for TI85 and TI86 at
https://www.ticalc.org/archives/files/fileinfo/31/3144.html
has load options for appmake executables but I get nothing after 
activating debug mode, loading and running. (I have found the roms for both the TI85 and TI86).

Am I using the wrong emulator?
Which emulator would you recommend for any of the TI systems?
If a shell is the only way to load them, how do I install the shell? Is it just a rom replacement?

zx70, 2018-05-24:
the preferred way is the shell, but I remember the ti86 is probably the weaker port.
I'd suggest to begin with ti85, most of the shells are a memory backup,  you should 'restore' them and, in some case,  activate the shell by executing a program you will find in the calculator.
the vti emulator is the correct choice,  but you need to learn how to transfer the files (the timeout happens soon and you must learn how to sinchronize the receive command and the file transfer, either via the right click menu or drag and drop)

Fabrizio-Caruso, 2018-05-24:
@zx70 I have the rigel.zip which does not contain any save state, but a .85B file... 
but no info on how to transfer it... If I try the obvious drag and drog or some transfer options, I get a transfer error.

Fabrizio-Caruso, 2018-05-24:
There is no clear option to load Rigel. 
Rigel's documentation talks about a TI-LINK cable, but I see no TI-LINK cable option in the emulator. 
I have tried:
1. loading the .85B as a save state
2. loading by drag and drop -> transfer error
3. loading as Send File to VTI and trying any of the options in : Enable Cable Link -> transfer error or it locks for ever
(I am using one of the suggested rom versions).

Maybe there is some hidden option to load Rigel.

Fabrizio-Caruso, 2018-05-24:
Starting the program under emulator for Windows PC
1. Install the emulator program VTI.
2. Download the TI85.ROM and put it into VTI directory.
3. Download and extract Rigel Shell.
4. Start VTI. Select TI85.
5. Click 2nd Link, then press F2 to receive.
6. Right Mouse click and choose Send File to VTI. Choose the file rigel.85b from the Backups directory of the Rigel Shell.
7. Quickly press F1 to confirm Warning about Memory backup, before transmission error message.
8. Click 2nd OFF On, to init the Rigel Shell.
9. Click 2nd Link, then press F2 to receive.
10. Right Mouse click and choose Send File to VTI. Select spice.85s.
11. Click 2nd OFF On, to init the Rigel Shell.
12. Click Custom, then F1, and select SPICE program.

Fabrizio-Caruso, 2018-05-24:
I am pretty sure I can get my game to work on the 5 Texas targets but I am not able to load one single binary.

zx70, 2018-05-24:
did you build the prophets with the proper start-up conf?

Fabrizio-Caruso, 2018-05-25:
@zx70 I have not even managed to install Rigel because I always get Transfer errors. I don't know how the the backup file is sent to VTI. I have tried multiple options.

I am compiling with no special options:
```
ti85:
	$(Z88DK_PATH)$(MYZ88DK) +ti85 \
	-O3 -D__TI85__ \
	-clib=ansi -pragma-define:ansicolumns=32 \
	-vn \
	-DFULL_GAME -DCLIB_ANSI \
	-lndos \
	-create-app -o $(BUILD_PATH)/FULL_ti85.bin  \
	$(SOURCE_PATH)/horizontal_missile.c $(SOURCE_PATH)/rocket.c $(SOURCE_PATH)/item.c \
	$(SOURCE_PATH)/display_macros.c  $(SOURCE_PATH)/enemy.c $(SOURCE_PATH)/invincible_enemy.c \
	$(SOURCE_PATH)/level.c $(SOURCE_PATH)/character.c \
	$(SOURCE_PATH)/text.c $(SOURCE_PATH)/missile.c $(SOURCE_PATH)/strategy.c $(SOURCE_PATH)/input_macros.c \
```

zx70, 2018-05-25:
you need a shell.
I just tried with Rigel and I succeded after a couple of attempts, the trick I used is to drag the file close to the F1 key and quickly clicking on F1 just after releasing the button to confirm the "memory backup" restore option.
You can try also with ZShell, (it should be compatible), but it won't help you to workaround the F1 problem.

Fabrizio-Caruso, 2018-05-25:
I have tried to:
drag the RIGEL.85B file exacly over the F1 button in VTI and I start clicking on it immediately but I still get the "Transmission Error". 

You may be using a different VTI version that allows you to load Rigel. Could you please tell which one you are using?

zx70, 2018-05-25:
Be patient, to upload a shell is very critical 
- reset the calculator and set it into RECEIVE mode (2nd, RECV)
- quickly enough drag the shell file onto the calculator
- very quickly, just after releasing the file, click on F1

if it fails, be patient.. reset the calculator and try again.

After you succeeded, exit the emulator and keep a copy of "ti85.sav"

Fabrizio-Caruso, 2018-05-25:
@zx70  there is no clear 2nd receive mode in the options. You may be using a different VTI version. I see lots of receive options some of which just freeze the calculator. The optuons I see are about different COM ports.

zx70, 2018-05-25:
I'm using Virtual TI 2.5.
[image]
Click on the three buttons respecting the 1,2,3 sequence

Fabrizio-Caruso, 2018-05-26:
Thanks! GRAZIE @zx70 !
I got it! Sorry for asking the detailed instructions!

closed 2018-05-26 by Fabrizio-Caruso