z88dk/z88dk #769: can't compile v1.99B on gentoo using gcc 6.4.0

rpolasek, 2018-05-25:
>>> Compiling source in /var/tmpfs/portage/dev-embedded/z88dk-1.99/work/z88dk-1.99 ...
echo '#define PREFIX "/usr/local/lib/z88dk"' > src/config.h
echo '#define UNIX 1' >> src/config.h
echo '#define Z88DK_VERSION "20180525"' >> src/config.h
make -C src/appmake
make[1]: Entering directory '/var/tmpfs/portage/dev-embedded/z88dk-1.99/work/z88dk-1.99/src/appmake'
gcc -march=haswell -O2 -pipe  -c appmake.c
gcc -march=haswell -O2 -pipe  -c z88.c
gcc -march=haswell -O2 -pipe  -c zxvgs.c
gcc -march=haswell -O2 -pipe  -c zx.c
gcc -march=haswell -O2 -pipe  -c z88shell.c
gcc -march=haswell -O2 -pipe  -c abc80.c
abc80.c: In function ‘datablockout’:
abc80.c:124:3: warning: ignoring return value of ‘fread’, declared with attribute warn_unused_result [-Wunused-result]
   fread(&buffer[3],253,1,fin);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~
gcc -march=haswell -O2 -pipe  -c zx81.c
gcc -march=haswell -O2 -pipe  -c msx.c
gcc -march=haswell -O2 -pipe  -c mtx.c
gcc -march=haswell -O2 -pipe  -c mz.c
gcc -march=haswell -O2 -pipe  -c nec.c
gcc -march=haswell -O2 -pipe  -c p2000.c
gcc -march=haswell -O2 -pipe  -c px.c
gcc -march=haswell -O2 -pipe  -c aquarius.c
gcc -march=haswell -O2 -pipe  -c c7420.c
gcc -march=haswell -O2 -pipe  -c rom.c
gcc -march=haswell -O2 -pipe  -c sorcerer.c
gcc -march=haswell -O2 -pipe  -c sos.c
gcc -march=haswell -O2 -pipe  -c svi.c
gcc -march=haswell -O2 -pipe  -c sc3000.c
gcc -march=haswell -O2 -pipe  -c ace-tap.c
gcc -march=haswell -O2 -pipe  -c hex.c
gcc -march=haswell -O2 -pipe  -c lynx.c
gcc -march=haswell -O2 -pipe  -c rex6000.c
gcc -march=haswell -O2 -pipe  -c tixx.c
tixx.c: In function ‘tixx_exec’:
tixx.c:202:5: warning: ignoring return value of ‘fread’, declared with attribute warn_unused_result [-Wunused-result]
     fread(buf, n, 1, fp);
     ^~~~~~~~~~~~~~~~~~~~
gcc -march=haswell -O2 -pipe  -c nascom.c
gcc -march=haswell -O2 -pipe  -c cpc.c
cpc.c: In function ‘cpc_exec’:
cpc.c:404:11: warning: ignoring return value of ‘fread’, declared with attribute warn_unused_result [-Wunused-result]
           fread(srchead, 128, 1, source);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cpc.c:469:16: warning: ignoring return value of ‘fread’, declared with attribute warn_unused_result [-Wunused-result]
                fread(srcdata, header[19] + header[20] * 256, 1, source);
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gcc -march=haswell -O2 -pipe  -c m5.c
gcc -march=haswell -O2 -pipe  -c mc.c
gcc -march=haswell -O2 -pipe  -c newbrain.c
gcc -march=haswell -O2 -pipe  -c newext.c
gcc -march=haswell -O2 -pipe  -c sms.c
sms.c: In function ‘sms_exec’:
sms.c:152:5: warning: ignoring return value of ‘fread’, declared with attribute warn_unused_result [-Wunused-result]
     fread(memory, sizeof(memory[0]), len, fpin);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gcc -march=haswell -O2 -pipe  -c trs80.c
gcc -march=haswell -O2 -pipe  -c c128.c
gcc -march=haswell -O2 -pipe  -c galaksija.c
gcc -march=haswell -O2 -pipe  -c vz.c
gcc -march=haswell -O2 -pipe  -c enterprise.c
gcc -march=haswell -O2 -pipe  -c x07.c
gcc -march=haswell -O2 -pipe  -c residos.c
gcc -march=haswell -O2 -pipe  -c inject.c
gcc -march=haswell -O2 -pipe  -c vg5k.c
gcc -march=haswell -O2 -pipe  -c z1013.c
gcc -march=haswell -O2 -pipe  -c extract.c
gcc -march=haswell -O2 -pipe  -c z9001.c
gcc -march=haswell -O2 -pipe  -c kc.c
gcc -o appmake -Wl,-O1 -Wl,--as-needed appmake.o z88.o zxvgs.o zx.o z88shell.o abc80.o zx81.o msx.o mtx.o mz.o nec.o p2000.o px.o aquarius.o c7420.o rom.o sorcerer.o sos.o svi.o sc3000.o ace-tap.o hex.o lynx.o rex6000.o tixx.o nascom.o cpc.o m5.o mc.o newbrain.o newext.o sms.o trs80.o c128.o galaksija.o vz.o enterprise.o x07.o residos.o inject.o vg5k.o z1013.o extract.o z9001.o kc.o
make[1]: Leaving directory '/var/tmpfs/portage/dev-embedded/z88dk-1.99/work/z88dk-1.99/src/appmake'
make -C src/appmake PREFIX=`pwd` install
make[1]: Entering directory '/var/tmpfs/portage/dev-embedded/z88dk-1.99/work/z88dk-1.99/src/appmake'
install appmake /var/tmpfs/portage/dev-embedded/z88dk-1.99/work/z88dk-1.99/bin
make[1]: Leaving directory '/var/tmpfs/portage/dev-embedded/z88dk-1.99/work/z88dk-1.99/src/appmake'
make -C src/copt
make[1]: Entering directory '/var/tmpfs/portage/dev-embedded/z88dk-1.99/work/z88dk-1.99/src/copt'
gcc -c -o copt.o -march=haswell -O2 -pipe -I. copt.c
gcc -c -o regex/regcomp.o -march=haswell -O2 -pipe -I. regex/regcomp.c
regex/regcomp.c: In function `regcomp':
regex/regcomp.c:104:29: error: `REG_DUMP' undeclared (first use in this function)
 # define GOODFLAGS(f) ((f)&~REG_DUMP)
                             ^
regex/regcomp.c:107:11: note: in expansion of macro `GOODFLAGS'
  cflags = GOODFLAGS(cflags);
           ^~~~~~~~~
regex/regcomp.c:104:29: note: each undeclared identifier is reported only once for each function it appears in
 # define GOODFLAGS(f) ((f)&~REG_DUMP)
                             ^
regex/regcomp.c:107:11: note: in expansion of macro `GOODFLAGS'
  cflags = GOODFLAGS(cflags);
           ^~~~~~~~~
regex/regcomp.c:108:39: error: `REG_NOSPEC' undeclared (first use in this function)
  if ((cflags&REG_EXTENDED) && (cflags&REG_NOSPEC))
                                       ^~~~~~~~~~
regex/regcomp.c:109:10: error: `REG_INVARG' undeclared (first use in this function)
   return(REG_INVARG);
          ^~~~~~~~~~
regex/regcomp.c:111:13: error: `REG_PEND' undeclared (first use in this function)
  if (cflags&REG_PEND) {
             ^~~~~~~~
regex/regcomp.c:112:11: error: `regex_t {aka struct re_pattern_buffer}' has no member named `re_endp'
   if (preg->re_endp < pattern)
           ^~
regex/regcomp.c:114:13: error: `regex_t {aka struct re_pattern_buffer}' has no member named `re_endp'
   len = preg->re_endp - pattern;
             ^~
regex/regcomp.c:176:6: error: `regex_t {aka struct re_pattern_buffer}' has no member named `re_g'
  preg->re_g = g;
      ^~
regex/regcomp.c:177:6: error: `regex_t {aka struct re_pattern_buffer}' has no member named `re_magic'
  preg->re_magic = MAGIC1;
      ^~
regex/regcomp.c:181:12: error: `REG_ASSERT' undeclared (first use in this function)
   SETERROR(REG_ASSERT);
            ^
regex/regcomp.c:58:32: note: in definition of macro ‘SETERROR’
 #define SETERROR(e) seterr(p, (e))
                                ^
regex/regcomp.c: In function ‘p_ere’:
regex/regcomp.c:210:27: error: ‘REG_EMPTY’ undeclared (first use in this function)
   REQUIRE(HERE() != conc, REG_EMPTY); /* require nonempty */
                           ^
regex/regcomp.c:58:32: note: in definition of macro ‘SETERROR’
 #define SETERROR(e) seterr(p, (e))
                                ^
regex/regcomp.c:210:3: note: in expansion of macro ‘REQUIRE’
   REQUIRE(HERE() != conc, REG_EMPTY); /* require nonempty */
   ^~~~~~~
regex/regcomp.c: In function ‘p_ere_exp’:
regex/regcomp.c:296:12: error: ‘REG_EMPTY’ undeclared (first use in this function)
   SETERROR(REG_EMPTY);
            ^
regex/regcomp.c:58:32: note: in definition of macro ‘SETERROR’
 #define SETERROR(e) seterr(p, (e))
                                ^
regex/regcomp.c: In function ‘p_str’:
regex/regcomp.c:393:18: error: ‘REG_EMPTY’ undeclared (first use in this function)
  REQUIRE(MORE(), REG_EMPTY);
                  ^
regex/regcomp.c:58:32: note: in definition of macro ‘SETERROR’
 #define SETERROR(e) seterr(p, (e))
                                ^
regex/regcomp.c:393:2: note: in expansion of macro ‘REQUIRE’
  REQUIRE(MORE(), REG_EMPTY);
  ^~~~~~~
regex/regcomp.c: In function ‘p_bre’:
regex/regcomp.c:436:27: error: ‘REG_EMPTY’ undeclared (first use in this function)
  REQUIRE(HERE() != start, REG_EMPTY); /* require nonempty */
                           ^
regex/regcomp.c:58:32: note: in definition of macro ‘SETERROR’
 #define SETERROR(e) seterr(p, (e))
                                ^
regex/regcomp.c:436:2: note: in expansion of macro ‘REQUIRE’
  REQUIRE(HERE() != start, REG_EMPTY); /* require nonempty */
  ^~~~~~~
regex/regcomp.c: In function ‘repeat’:
regex/regcomp.c:993:12: error: ‘REG_ASSERT’ undeclared (first use in this function)
   SETERROR(REG_ASSERT); /* just in case */
            ^
regex/regcomp.c:58:32: note: in definition of macro ‘SETERROR’
 #define SETERROR(e) seterr(p, (e))
                                ^
make[1]: *** [Makefile:18: regex/regcomp.o] Error 1

suborb, 2018-05-25:
This looks like the compile of copt is missing the -DLOCAL_REGEXP flags. 

We did a bit of a reorg of code since that release so I suspect it's already fixed.

rpolasek, 2018-05-28:
unfortunately, v1.99B is the latest release. do you plan to release next version in near future?

aralbrec, 2018-05-28:
We've been planning to do one for a while but you can also get the current version from here or from the [nightly build z88dk-latest.tgz](http://nightly.z88dk.org/).

Some install instructions:
https://www.z88dk.org/wiki/doku.php?id=temp:front#linux_unix

Some instructions for compiling zsdcc:
https://github.com/z88dk/z88dk/tree/master/src/zsdcc

feilipu, 2019-07-04:
Closed because really old, and source is not current.
Try with the current code base and reopen if it still doesn't work, please.

closed 2019-07-04 by feilipu