z88dk/z88dk #770: zsdcc: invalid preserves_regs list causes unexpected behaviour

aralbrec, 2018-05-25:
I'm just copying this report here to check on later.

> I was originally specifying ixh and ixl in a __preserves_regs call, but on inspection it caused the calling assembly to ignore *all* the registers in __preserves_regs. I.e. with no __preserves_regs I could see the caller pushing and popping DE, with __preserves_regs(d,e,iyh,iyl) it doesn't but with __preserves_regs(d,e,ixh,ixl,iyh,iyl) it goes back to pushing and popping DE. 

Specifying ixl,ixh in the preserves _regs list is invalid since sdcc needs to use ix as frame pointer and it expects code to save it if it does get used.

spth, 2018-05-26:
Is this a silent failure in zsdcc? Upstream SDCC emits a warning when it encounters an invalid entry (and stops processing the arguments at that point - any registers specified after the invalid one will be ignored).

Philipp

aralbrec, 2018-05-28:
I've tried it myself and it is not silent:

zzz.c
```
extern void foo(unsigned int) __preserves_regs(a,b,c,d,e,h,l,iyl,iyh,ixl,ixh);

void main(void)
{
	for (unsigned int i = 0; i != 0xc000; i += 0x10)
		foo(i);
}
```

zcc +zxn -vn -a -clib=sdcc_iy -SO3 zzz.c

```
zzz.c:1: warning 235: unknown register specification
```

But if the register specification is unknown, it is thrown out entirely:

Invalid with ixl,ixh, there is an extra `push bc` around the call to foo.
```
l_main_00116:
	push	bc
	push	bc
	call	_foo
	pop	af
	pop	bc
```

Valid with ixl,ixh removed, the extra push is gone:

```
l_main_00116:
	push	bc
	call	_foo
	pop	af
```

I think the user was expecting an error if the register specification was going to be thrown out rather than a warning which it looks like he missed.  Maybe for a warning, the register specification should be best effort rather than thrown out but I am of two minds on that.

spth, 2018-05-28:
I now changed it (in upstream SDCC, I have no idea when the change will arrive in zsdcc) so
1) The unknown register is mentioned in the error message
2) All known registers are still considered for optimization

Philipp

feilipu, 2019-02-11:
Obsolete? Fixed in sdcc.

closed 2019-02-11 by suborb