z88dk/z88dk #771: [einstein][bug] Frozen during load (it somehow depends on ram used or the binary size)

Fabrizio-Caruso, 2018-05-25:
I am now testing the Tatung Einsten with the snapshot from 2018-05-23.

Depending on ram used or the binary size: 
1. It fails if I set a non-default column size (but the default one is a nice one due to color clash)
2. It fails if I set my constant XSize to 40 (XSize sets the number of horizontal tiles that should be 64 or 80 with the default settings)

I compile with:
```
einstein_tiny:
	$(Z88DK_PATH)$(MYZ88DK) +cpm -leinstein \
	-D__EINSTEIN__ \
	-DFORCE_CONIO \
	-DTINY_GAME -DLESS_TEXT -DNO_SLEEP -DNO_WAIT \
	-clib=ansi \
	-create-app -o$(BUILD_PATH)/TINY_einstein.bin \
	$(SOURCE_PATH)/display_macros.c \
	$(SOURCE_PATH)/enemy.c \
	$(SOURCE_PATH)/level.c $(SOURCE_PATH)/character.c $(SOURCE_PATH)/text.c  \
	$(SOURCE_PATH)/strategy.c $(SOURCE_PATH)/input_macros.c \
	$(SOURCE_PATH)/main.c
	rm $(BUILD_PATH)/TINY_einstein.bin
```

suborb, 2018-05-30:
The Einstein is a tms9118 machine so it should be fairly standard 32x24 - though I’ve not enabled the generic console there. 

The Einstein 256 had an evolution on the chip and offered a better resolution. 

If you can point me in the direction of Roms etc i’ll get the Einstein up to the same levels as the other tms machines.

Fabrizio-Caruso, 2018-08-28:
It looks much better now. I am going to test it a bit more but the issue may have been solved somehow.

closed 2018-09-03 by suborb