z88dk/z88dk #772: [einstein][improvement][keyboard] Keys kept pressed do not produce multiple detections

Fabrizio-Caruso, 2018-05-25:
The Einstein target has a little input issue: keys kept pressed do not produce multiple detections.

suborb, 2018-05-30:
If that’s the case then we’ll have to write an inkey driver for it. 

Should be possible to figure out the ports and bits from the mess source code. 

Since inkey is my current specialised subject I’ll pick this up after the vg5k work.

Fabrizio-Caruso, 2018-08-28:
@suborb the einstein target seems more stable than last time I tested it.
The input issue still remains.

By the way, in order to test it, I always have to go through a painful manual procedure with EDIP 1e. Do you do something smarter/faster?

Fabrizio-Caruso, 2018-08-28:
@suborb 
MAME/MESS 0.196 emulates it well enough.

There may be a smarter way to test it but I go through many steps:
USE EDIP1 1e (http://www.einstein.talktalk.net/edip.html?showpage=true)
to insert the .COM file into a disk image:
0. If necessary. rename your file such that no special characters (no underscore) are present
1. Run EDIP 1e
2. Select DOS80.DSK
3. Click on CREATE BLANK -> Name your disk image
4. Select DOS TATUNG XTAL...
5. Press SELECT -> Select File
6. Press SELECT -> Cancel -> Quit

Use MAME/MESS:
1. mame64.exe einstein -window -uimodekey DEL -flop1 c:\Emulation\Mame\EDIP1e\XCHASE.DSK
2. run the .COM file by typing its name

suborb, 2018-08-28:
Thanks, I have something working now.

It looks like a fairly standard CP/M disc (possibly one of the CPC formats?) - albeit one with some stuff on the boot sector - have you tried cpmtools on it? -

Fabrizio-Caruso, 2018-08-28:
@suborb I tried the same cpc tool I use for the cpc target but it did not work for me.
I don't know if and how to script cpmtools so that everything is automatic. Is it possible?

suborb, 2018-08-28:
For my own sake I’m going to need to write a tool to do this - edip is awful to use. 

It’s on the list :)

suborb, 2018-08-30:
@Fabrizio-Caruso try +cpm -subtype=einstein

A .dsk will be produced that can be used in combination with dos80.dsk.

-flop1 dos80.dsk -flop2 [created image]

Once booted, type:

```
1:
APP.COM
```

And hopefully your program will start.