z88dk/z88dk #774: [einstein][appmake][question] The Tatung Einstein may use the same dsk as CPC

Fabrizio-Caruso, 2018-05-25:
While creating disk images for the Tatung Einstein with the recommended tool EDIP1E I see that among the formats there is the Extended CPC DSK, which seems to work sometimes (this target freezes most times during the program load).

If things are as they seem, we may get ready-to use disk images for the Tatung Einstein by:
1. compile the einstein target to produce a raw binary
2. run cpc app-make on the einstein binary to add the required header (in general it may be useful to have a way to select an appmake implementation from a different target)
3. use  cpcxfsw to insert the file with header into a disk image as I do for the cpc (see below)

```
...
	$(TOOLS_PATH)/cpcxfsw -nd FULL_cpc.dsk
	$(TOOLS_PATH)/cpcxfsw FULL_cpc.dsk -p build/FULL_cpc.cpc xchase
```

zx70, 2018-05-29:
it seems a hint rather than a bug,  what are you asking exactly?

Fabrizio-Caruso, 2018-05-29:
@zx70  indeed it is not bug. It is a hint and I am not sure it is a good one.
I just saw "Extended CPC DSK" as one of the options in EDIP1E and this made me think it the same disk format and appmake as CPC could be used.

Fabrizio-Caruso, 2018-05-30:
I am closing this non-issue.
It is just a hint for a possible future feature (if my hint is right).

closed 2018-05-30 by Fabrizio-Caruso