z88dk/z88dk #776: [ti86][question] How to load assembly? Ion86? Direct assembly load without a shell?

Fabrizio-Caruso, 2018-05-27:
I am trying to load my game on an emulated TI86 (Is VTI good for this target?)
I read that ION is the default shell but others are possible and even direct assembly.

Can I compile my game in a way that can be loaded without any shell?
I have read that the TI86 should be capable of this.

Which shell exactly should I use?
I cannot find any other Ion for the TI86 other than Ion86 (a Ti83 emulator)
https://www.ticalc.org/archives/files/fileinfo/109/10997.html

Ion86 is not distributed as a memory backup but just as a .86p file. How do I load it?
Using the 2ND X-VAR (LINK) F2 and dragging the .86p fails with "Transmission Error".

If I try to compile my game with any other startup option I get tons of warnings:
```
...
Warning at file '../_DEVELOPMENT/error/z80/error_llmc.asm' line 17: integer '68910' out of range
Warning at file '../_DEVELOPMENT/error/z80/error_llmc.asm' line 19: integer '68910' out of range
Warning at file '../_DEVELOPMENT/error/z80/error_lmc.asm' line 17: integer '68895' out of range
Warning at file '../_DEVELOPMENT/z80/z80/asm_z80_delay_ms.asm' line 42: integer '69084' out of range
Warning at file '../_DEVELOPMENT/z80/z80/asm_z80_delay_ms.asm' line 51: integer '69084' out of range
```

suborb, 2018-05-27:
I can't answer any other bits, but those warnings mean that your program is too long usually. Programs start at 0xD748 for the ti86 which gives you around about 10k.

Fabrizio-Caruso, 2018-05-27:
@suborb I do not get the error with the default startup.

My main problem for me is understanding how to load the binary. I am struggling to find somewhere how this is done.

aralbrec, 2018-05-28:
> Warning at file '../_DEVELOPMENT/error/z80/error_llmc.asm' line 17: integer '68910' out of range
> Warning at file '../_DEVELOPMENT/error/z80/error_llmc.asm' line 19: integer '68910' out of range [...]

The warnings do indicate that your program is too big but there's also something else here.  "error_llmc" is an error entry point for 64-bit integers and "error_lmc" is an error entry point for 32-bit integers.  So I think you are doing a zsdcc compile but somehow you are getting additional code you're not using or wanting.  This might happen if you're using, eg, printf misconfigured to support 32/64-bit integer printing.

The classic library should auto-detect a printf subset you're using but you can also specify which printf converters you want with a pragma like `#pragma printf = "%0d %s ..."` (for classic "%0d" supports formatting and is more complete than "%d").  

Anyway you can check what printf subset is being used by looking at the file zcc_opts.def after a compile.  There will be CLIB_OPT_PRINTF and CLIB_OPT_PRINTF2 variables defined that will indicate what is going into printf.

Fabrizio-Caruso, 2018-09-21:
@zx70  I would like to start with something small and simple.
How can I just get a simple "hello world" example (or the tiny/light version of my game) into the calculator?
I am not able to do it with VTI.
Could you please tell which steps I should follow?

Fabrizio-Caruso, 2018-09-21:
@zx70 The doc suggests Ion for the TI-86.
I cannot even get Ion86 to start.
I can drag Ion86 and I can drag my game.
If I  press on PRGM I see NAMES.
If I select NAMES I see A and Ion86 but trying to start any of the two will produce
ERROR 07 SYNTAX

Fabrizio-Caruso, 2018-09-24:
@zx70  I confirm that with Wabbitemu I can start the binary and I get a crash at startup.
I have tried both the default startup and the plain asm startup with and without the suggest Ion86 shell and I always get the same result.

Fabrizio-Caruso, 2018-09-24:
@zx70, indeed LargeLd works and the binary can be loaded and it runs more or less as expected.
I was confused by the fact that the doc seems to suggest the Ion shell.

closed 2018-09-24 by Fabrizio-Caruso