z88dk/z88dk #778: [ti83] Shell seems OK but ERR:UNDEFINED when selecting the transfered binary

Fabrizio-Caruso, 2018-05-28:
I have managed to install ion.1.5 into VTI for the ti83 with the latest rom.

I can also transfer the .83p binary by:
2nd LINK ENTER -> List of items starting with A, ION, IONZ, etc...
Should I select an item?
I then drop the .83p binary -> I get the same message:
"prgmION 
Ion Installed
Done"
I click on PRGM and I see now my binary in the list.
I select the binary and I get 
ERR:UNDEFINED

zx70, 2018-05-28:
Shouldn't you start the shell  ?
You probably see the file also as.. a file but the TI calculator can't handle it directy.

Fabrizio-Caruso, 2018-05-28:
What do you mean by "start the shell"?
I can even load the game. It will just produce an error when run but it does appear in the list.

zx70, 2018-05-28:
You cannot ask the calculator to run the game, you should rather start the shell and run the game from within the shell.
So:
- Copy ION.83G  (drag it onto VTI, no need to ask to transfer(
- run the ION installer (type "prgmION" directly or use the PRGM key to choose it from the program list)
- copy your program onto the calculator (be careful to avoid confusion and avoid the default program name, the ION shell entry name is 'A' !)
- run the shell (type "prgmA" directly or use the PRGM key to choose it from the program list), the shell won't start if no valid programs are in memory,
- use the shell to choose the program and run it.

Fabrizio-Caruso, 2018-05-28:
@zx70  Thanks, in the doc that comes with the Ion shell I read:
"
>Installation
Send ion.8xg to your TI-83 Plus or
Send ion.83g to your TI-83.
To send to your calculator using the Graph-Link
software, select [Link|Send To|RAM] for the
TI-83 Plus or [Link|Send] for the TI-83.
Run prgmION.
"
I have followed this instructions.
Now I will try to run a instead.

Fabrizio-Caruso, 2018-05-28:
I may have figured it out. I was selecting receive to receive the file. Now I have tried send and it sort of worked...

Fabrizio-Caruso, 2018-05-28:
I have figure it out... Pressing quickly return I managed to avoid the "Trasmission Error"...

closed 2018-05-28 by Fabrizio-Caruso