z88dk/z88dk #782: [sorcerer] VT52? How? Sequences seem not to work

Fabrizio-Caruso, 2018-05-30:
VT52 sequences do not seem to work in my games if I use redefined conio APIs as:
```
	#define gotoxy(x,y) printf("\033Y%c%c",y+31,x+31)
	#define clrscr() printf("\033H\033J")
	#define cprintf printf
	#define cputc(c) fputc_cons(c)
```
I have compiled it with:
```
srr:
	$(Z88DK_PATH)$(MYZ88DK) +srr -O3 -pragma-redirect:fputc_cons=fputc_cons_native \
	-D__SRR__ -vn \
	-DFULL_GAME -DSOUNDS \
	-DEND_SCREEN -DBETWEEN_LEVEL \
	-DCONIO_VT52 \
	-lndos \
	-create-app -o $(BUILD_PATH)/FULL_srr.prg  \
...
```

[image]

suborb, 2018-05-30:
I’m afk but if I didn’t enable it by default you need to add -pragma-redirect:fputc_cons=fputc_cons_generic

Fabrizio-Caruso, 2018-05-30:
@suborb , as you can see in my example above, I have, but no sequences are recognized.

Fabrizio-Caruso, 2018-05-30:
@suborb I still need to write an isolated test, though.

suborb, 2018-05-30:
You’ve got _native not _generic in your compile line.

Fabrizio-Caruso, 2018-05-30:
@suborb, you are just right as usual! Sorry and THANKS! 
I may open a different issue on the input. I am going to test it thoroughly.

suborb, 2018-05-30:
I’ll look into that cls problem. It should work since my test calls cls, but not using those codes. The clrscr() macro in conio.h uses the method I used. 

The x,y should be +32 btw

Fabrizio-Caruso, 2018-05-31:
@suborb THANKS! 
The "\033H\033J" sequence seems to work as HOME + J, i.e., 'J' is displayed at the top left corner.
I confirm the issue even with a simple isolated test here (unless I am using the wrong sequences):
```
#include <stdio.h>
#define gotoxy(x,y) printf("\033Y%c%c",y+32,x+32)
#define clrscr() printf("\033H\033J")
#define cprintf printf
int main()
{
	gotoxy(10,10);
	cprintf("Hello world!\n");	
	getch();
	clrscr();
	gotoxy(12,12);
	cprintf("Hello world!\n");	
}
```

suborb, 2018-05-31:
Ah, that's deliberate, these two:

```
IF SUPPORT_vt52x
        cp      'K'
        jr      z,clear_eol
        cp      'J'
        jr      z,clear_eos
ENDIF
```

aren't compiled in by default, just use \033E which will clear the screen and go to home (and is quicker in anycase)

closed 2018-06-01 by Fabrizio-Caruso