z88dk/z88dk #787: [sharp x1] adding graphics and general improvements

zx70, 2018-06-01:
I just succeeded in decoding the core entry table of the HuBASIC interpreter and put the results in the basck.c tool,  which greatly helps in disassembling all its variants .
It will unveil lots of useful information on the Sharp machines,  so far I tested it on the mz700 basic and the x1 disk image, I'll try with the mz2500 too.

zx70, 2018-06-15:
I put the result of my investigations in the "techdocs" project, (X1 folder).    Do you think I can add also my derived source of MSX (it can be assembled to produce the original ROM) and other similar stuff ?

suborb, 2018-09-01:
@zx70 did you figure out how to access the pcg memory? The font on the x1 is awful and it would be nice to be able to reconfigure it.

zx70, 2018-09-01:
I could have something more somewhere,  but in the mesntime you can look fot 'def chr$' in my basic disassembly:  https://github.com/z88dk/techdocs/blob/master/targets/x1/hubasic-CZ8FB01.asm

suborb, 2019-11-10:
Time to do something on the X1, it's a capable machine, lets make it better:

- [x] Gencon to support switching between 40 and 80 column mode
- [x] Appmake to generate CP/M discs
- [ ] Get mode switching working in takeda and mame
- [ ] Make vt100 handle dynamic mode switching 
- [ ] Gencon to support PCG - custom fonts/UDGs
- [ ] 160x50, 80x50 lores graphics (with PCG support) or very lores 40x25, 80x25
- [ ] Hires graphics (overlaid on text - port 0x1300 sets above/below/mixed)

I think my font issues were caused by a bad dump - I've taken the mame files and it's a lot better now.

suborb, 2019-11-12:
I've just about got the hires graphics working now after a fair bit of head scratching (point is still broken at the moment) - as usual I forgot to set the palette so nothing was being shown on screen.

The hires graphics are in rgb planes starting at $4000,$8000,$c000 in the IO space. The screen is split into characters, so that the (0,0) character location has (when using 8x8 characters), the following addresses in the blue plane:

$4000 - pixel row 0
$4800 - pixel row 1
$5000 - pixel row 2
$5800 - pixel row 3
$6000 - pixel row 4
$6800 - pixel row 5
$7000 - pixel row 6
$7800 - pixel row 7

Location (0,1) will have pixel row 0 at $4001. Location (1,0) in 40 column mode starts at $4028, in 80 column mode $4050

zx70, 2019-11-13:
this is great news,  I can't wait to try building some of our examples on it!  there is something in the microbee library used to get hrg with a possibly similar trick,  which maybe is worth to try porting to the x1.

suborb, 2019-11-13:
I've almost got the PCG working, bar an issue with the space character and a difference in behaviour between mame and takeda:

[image]

Loading the PCG is a "little" slow and needs the text VRAM to be populated - hence why a lot of the games for the X1 pause with black screens/"please wait" screens - effectively only one character can be redefined every 3 frames.

suborb, 2019-11-14:
Final PR about to be posted:

- CPM library containing, gencon, ansi, graphics, psg, input
- Appmake using the disc utilities now creates a d88 files
- Appmake can now create discs containing apps compiled with -subtype=im2
- crt0 refactored to disentangle allram and ipl modes
- All x1 code is now in target/x1 
- Interrupt handlers available in allram mode

I think this target is now sorted, so I'm going to shuffle over to the SMC-777 and give it the same treatment.

PR #1328 (2019-11-14): (x1/classic) Final bit of rework
Fix #787 

- CPM library containing, gencon, ansi, graphics, psg, input
- Appmake using the disc utilities and now creates a d88 files
- Appmake can now create discs containing apps compiled with -subtype=im2
- crt0 refactored to disentangle allram and ipl modes
- All x1 code is now in target/x1
- Interrupt handlers available in allram mode
closed 2019-11-14 by suborb