z88dk/z88dk #788: [eg2k][question/bug] The suggested Genieous emulator does not seem to load .cmd files

Fabrizio-Caruso, 2018-06-01:
I can produce disk images but how can I load them?
I am using the suggested emulator: Genieous but nothing happens if I try to load the .cmd file.

zx70, 2018-06-02:
I used it recently, it worked :(

suborb, 2018-06-04:
The easiest way to load into this emulator is to compile as follows:

    zcc +trs80 [files] -subtype=eg2000disk -create-app

Then, load up Genieous, go to `File->Autostart File`. Select your .cmd and it'll load and start automatically.

Fabrizio-Caruso, 2018-06-05:
Of course, I used +trs80 -subtyoe=eg2000disk -create-app:
``` 
	$(Z88DK_PATH)$(MYZ88DK) +trs80 \
	-subtype=eg2000disk \
	-lndos \
	-lm \
	-create-app \
...
	mv a.cmd $(BUILD_PATH)/FULL_eg2k.cmd
```

If I try to load the .cmd with autostart, I only get a blinking cursor.

Fabrizio-Caruso, 2018-06-05:
@suborb, the very same code (not just meta-code) runs on the trs80.

Fabrizio-Caruso, 2018-06-05:
@suborb The problem is probably caused by the fact that trs80 code does not run on the EG2K.
For the trs80, I assume it has memory mapped video ram at BASE_ADDR (unsigned short)(0x3c00).
I write directly into video ram to do my graphics.
I guess that the two systems are not identical.

suborb, 2018-06-05:
That will be it. There’s an assembler variable (base_graphics) which is set up by the crt to be the start of the graphics memory. You should be able to get hold of it with a short assembler routine.

Fabrizio-Caruso, 2018-06-05:
@suborb, can't I simply use base_graphics in my C code? How? 
I could simply define my BASE_ADDR differently for the EG2000, if I knew where the video memory is mapped on this machines.

suborb, 2018-06-05:
It's an internal variable used by the graphics subsystem - it came about because the z88 display can actually be anywhere in memory. It may not normally be set to anything sensible, however the trs80/eg2000 does set it in the crt0.

You'll need to write a function along the lines of:

    void *get_base_graphics() __naked
    {
    #asm
         EXTERN base_graphics
         ld hl,(base_graphics)
         ret
    #endasm
    }

The generic console should be pretty quick to be honest - I think you had to do vram writes previously because there used to be no way of changing the "cursor" position.