z88dk/z88dk #789: [sprinter][g800] Generic console broken with examples

Fabrizio-Caruso, 2018-06-02:
The generic console is broke in pps (sprinter) and g800 target 
I compile the pps target (and similarly the g800 target) with:
```
pps_vt52_test:
	$(Z88DK_PATH)$(MYZ88DK) +pps -O3 \
	-pragma-redirect:fputc_cons=fputc_cons_generic \
	$(SOURCE_PATH)/../experiments/vt52_test.c \
	-lndos -vn  \
```
I get:
```
Error at file 'c:/Z88DK/lib/../lib/crt/classic/crt_runtime_selection.asm' line 332: symbol 'DEFINED_fputc_cons' already defined
Error at file 'c:/Z88DK/lib/../lib/crt/classic/crt_runtime_selection.asm' line 334: symbol 'fputc_cons' already defined
2 errors occurred during assembly
Errors in source file c:\Z88DK\lib\config\..\..\\lib\target\pps\classic\pps_crt0.asm:
Error at file 'c:/Z88DK/lib/../lib/crt/classic/crt_runtime_selection.asm' line 332: symbol 'DEFINED_fputc_cons' already defined
                   ^ ----       defc DEFINED_fputc_cons = 1
Error at file 'c:/Z88DK/lib/../lib/crt/classic/crt_runtime_selection.asm' line 334: symbol 'fputc_cons' already defined
                   ^ ----       defc fputc_cons = fputc_cons_ansi
make: *** [Makefile:2241: pps] Error 1
```

test code:
```
#include <stdio.h>
#define gotoxy(x,y) printf("\033Y%c%c",y+32,x+32)
#define clrscr() printf("\033H\033J")
#define cprintf printf
int main()
{
	gotoxy(10,10);
	cprintf("Hello world!\n");	
	getch();
	clrscr();
	gotoxy(12,12);
	cprintf("Hello world!\n");	
}
```

suborb, 2018-06-02:
I haven’t added anything for g800 so that’s still on the ansi terminal by default.

closed 2018-06-02 by suborb

Fabrizio-Caruso, 2018-06-02:
@suborb Thanks! I saw VT52 support for g800 in the list of targets.

zx70, 2018-10-09:
Thanks to @ashitani graphics support is now available !
I'm linking in the standard basic graphics support (simplified putsprite etc) right now, so a good portion of the graphics example will be functional by tomorrow morning.