z88dk/z88dk #790: [sc3000][getk_inkey] broken in latest snapshot

Fabrizio-Caruso, 2018-06-02:
It seems that the latest snapshot or the one just before has broken sc3000.

If I compile (previously working):
```
sc3000_32k:
	$(Z88DK_PATH)$(MYZ88DK) +sc3000 \
	-O3 \
	-DFULL_GAME -DEND_SCREEN -DBETWEEN_LEVEL \
	-clib=ansi \
	-pragma-define:ansicolumns=32 \
	-vn -lndos -create-app -Cz--audio \
	-o $(BUILD_PATH)/FULL_sc3000_32k.prg \
...
```
I now get:
```
Error at file 'stdio/inkey/getk.asm' line 8: symbol 'getk_inkey' not defined
Error at file 'stdio/inkey/getk.asm' line 9: symbol 'getk_inkey' not defined
Error at file 'stdio/inkey/fgetc_cons.asm' line 8: symbol 'fgetc_cons_inkey' not defined
Error at file 'stdio/inkey/fgetc_cons.asm' line 9: symbol 'fgetc_cons_inkey' not defined
4 errors occurred during assembly
Errors in source file c:\Z88DK\lib\config\..\..\\lib\target\sc3000\classic\sc3000_crt0.asm:
Error at file 'stdio/inkey/getk.asm' line 8: symbol 'getk_inkey' not defined
Error at file 'stdio/inkey/getk.asm' line 9: symbol 'getk_inkey' not defined
Error at file 'stdio/inkey/fgetc_cons.asm' line 8: symbol 'fgetc_cons_inkey' not defined
Error at file 'stdio/inkey/fgetc_cons.asm' line 9: symbol 'fgetc_cons_inkey' not defined
```

Fabrizio-Caruso, 2018-06-03:
@suborb multiple targets do not longer compile and compilation produces
a 'getk_inkey' not defined error

This issue has just appeared.

suborb, 2018-06-03:
I think I’ve forgotten to check something in. I’m back tomorrow so I’ll take a look on Monday evening. 

Sorry about that.

suborb, 2018-06-04:
There was a missing file input/input_keyboard.lst which contained the appropriate machines.

I've now committed it, and raised #795 to make the build fail!

closed 2018-06-04 by suborb