z88dk/z88dk #791: [integration tests][feature request] Could it be to have some simple automatic integration tests that are after each pull-request and/or merge?

Fabrizio-Caruso, 2018-06-02:
I am seeing lots of targets fail to compile (vt52 on different targets, sc3000 broken,...).

I would like to help if I can. I could provide some simple compilation tests to check things like conio, vt52, vt100, sprites, etc...
Once I am done with the clean-up of CROSS CHASE, I will provide it a check but it is loooong to compile test. Maybe for automatic checks, something simpler could be used. 

Some of these issues could be prevented and detected automatically by a simple compilation of a batch of "hello world" tests.

Could it be to have some simple automatic integration tests that are after each pull-request and/or merge?

suborb, 2018-06-02:
So as past of the build we:

- compile hello world with the default config
- compile hello world with ansiterminal

I haven’t yet added generic console. 

It might be worth adding a fgetc test in there as well.

suborb, 2018-06-04:
I've added:

* world.c now has a fgetc_cons() in it
* world_conio.c now uses conio functions and kbhit()/getch()

closed 2018-06-04 by suborb