z88dk/z88dk #794: [msx] broken but compiles

Fabrizio-Caruso, 2018-06-03:
It seems that the msx target is broken.

If I compile my game I get no error but the executable just produces a black screen.

suborb, 2018-06-04:
@Fabrizio-Caruso I've just tried out adv_a.c with the following compile lines:

    zcc +msx adv_a.c -create-app -lndos
    zcc +msx adv_a.c -create-app -lndos -pragma-redirect:fputc_cons=fputc_cons_redirect
    zcc +msx adv_a.c -create-app -lndos -clib=ansi

So all 3 consoles and all 3 configurations display text.

Fabrizio-Caruso, 2018-06-04:
I confirm that with today's snapshot I still just get a black screen.

I will need to isolate this problem that has first appeared on the 1st of June (29th of May was OK).

Fabrizio-Caruso, 2018-06-04:
@suborb, @zx70 , I have found the portion of the code that is no longer behaving as until the 29th of May.
The following code is supposed to select general colors, mode and colors for groups of 8 characters:
```
	set_color(15, 1, 1);
        set_mode(mode_1);
	msx_vpoke(8192+ 0, 9*16); // 
	msx_vpoke(8192+ 1, 7*16); // 
	msx_vpoke(8192+ 2, 9*16); // 
	msx_vpoke(8192+ 3, 11*16); // 
	msx_vpoke(8192+ 4,10*16); // White !, $ -- 32 - 39
	msx_vpoke(8192+ 5, 2*16); // Green  -- 40 - 47
	msx_vpoke(8192+ 6, 4*16); // Green -- 48 - 55
	msx_vpoke(8192+ 7, 4*16); // Green -- 56 - 63
	msx_vpoke(8192+ 8, 8*16); // Red 64 --
	msx_vpoke(8192+ 9, 8*16); // Red 72 --
	msx_vpoke(8192+10, 8*16); // Red 80 --
	msx_vpoke(8192+11, 8*16); // Red 88 --
```

suborb, 2018-06-04:
I've made some fairly major TMS9918 changes last week - in particular set_mode() was changed to control the hardware directly rather than the through the bios.

Fabrizio-Caruso, 2018-06-04:
@zx70 has changed how vpoke interprets addresses but I think that he has tried to make the video memory always appear as on the MSX. So, if the code is correct, my portion of code should still work as before.

suborb, 2018-06-04:
I've switched back setting the screen via bios on the +msx. I've just tested cross chase and we get a display.

I'll do some work on mode 1 support then :)

closed 2018-06-04 by suborb