z88dk/z88dk #803: [einstein] Broken compilation: symbol 'msx_attr' not defined

Fabrizio-Caruso, 2018-06-07:
I am now getting these errors:
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 29: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 65: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 67: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 72: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 74: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 91: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 100: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 108: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 130: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 133: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 151: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 154: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_char.asm' line 382: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_cls_nobios.asm' line 33: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_cls_nobios.asm' line 40: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_cls_nobios.asm' line 55: symbol 'msx_attr' not defined
16 errors occurred during assembly
Errors in source file c:\Z88DK\lib\config\..\..\\lib\target\cpm\classic\cpm_crt0.asm:
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 29: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 65: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 67: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 72: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 74: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 91: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 100: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 108: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 130: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 133: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 151: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_attr.asm' line 154: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_char.asm' line 382: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_cls_nobios.asm' line 33: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_cls_nobios.asm' line 40: symbol 'msx_attr' not defined
Error at file 'stdio/ansi/msx/f_ansi_cls_nobios.asm' line 55: symbol 'msx_attr' not defined
make: *** [Makefile:1639: einstein] Error 1

suborb, 2018-06-07:
Argh, I deliberately didn't add generic console to Einstein since I've not got around to testing it.

But when I shared variables it becomes needed....so I've now added it. Sorry

closed 2018-06-07 by suborb

Fabrizio-Caruso, 2018-06-07:
@suborb testing the Einstein is painful due to the fact that you need to use an user-unfriendly tool and to the fact that it is a nasty bugs that makes it crash during load most of the times.

Fabrizio-Caruso, 2018-06-07:
Cool! You don't like Mame? 
By the way I thought the binaries for the Tatung Einstein could be stored into a cpc disk image.
I tried and I failed. Maybe I will have to use a hex editor to try to reverse engineer how the disk images work and maybe figure out how to create them without the horrible EDIP1E.

commit bc8d4a70 (2018-06-07): Fix #803: Include generic console in einstein for the variables
Fix #804: Get the name of the interrupt file right for sg1000