z88dk/z88dk #804: [sg1000][sc3000 -subtype=rom] cannot read file 'crt/classic/tms/interrupt.asm'

Fabrizio-Caruso, 2018-06-07:
Compiling for the SG1000 (SC3000 -subtype=rom) produces
"cannot read file 'crt/classic/tms/interrupt.asm'"

```
/cygdrive/c/z88dk/bin/zcc.exe +sc3000 -subtype=rom \
-O3 \
-DFULL_GAME -DEND_SCREEN -DBETWEEN_LEVEL \
-clib=ansi \
-pragma-define:ansicolumns=32 \
-vn -lndos -create-app \
-o build/FULL_sg1000.prg \
src/horizontal_missile.c src/rocket.c src/item.c src/end_screen.c \
src/missile.c src/invincible_enemy.c \
src/display_macros.c src/enemy.c src/level.c src/character.c src/text.c \
src/strategy.c src/input_macros.c src/main.c
Error at file 'c:/Z88DK/lib/../lib/target/sc3000/classic/rom.asm' line 70: cannot read file 'crt/classic/tms/interrupt.asm'
Error at file 'c:/Z88DK/lib/../lib/target/sc3000/classic/rom.asm' line 90: cannot read file 'crt/classic/tms/interrupt_handler.asm'
Error at file 'c:/Z88DK/lib/../lib/target/sc3000/classic/rom.asm' line 86: symbol 'int_handler' not defined
3 errors occurred during assembly
Errors in source file c:\Z88DK\lib\config\..\..\\lib\target\sc3000\classic\sc3000_crt0.asm:
Error at file 'c:/Z88DK/lib/../lib/target/sc3000/classic/rom.asm' line 70: cannot read file 'crt/classic/tms/interrupt.asm'
                   ^ ----       INCLUDE "crt/classic/tms/interrupt.asm"
Error at file 'c:/Z88DK/lib/../lib/target/sc3000/classic/rom.asm' line 90: cannot read file 'crt/classic/tms/interrupt_handler.asm'
                   ^ ----       INCLUDE "crt/classic/tms/interrupt_handler.asm"
Error at file 'c:/Z88DK/lib/../lib/target/sc3000/classic/rom.asm' line 86: symbol 'int_handler' not defined
                   ^ ----       call    int_handler
make: *** [Makefile:993: sg1000] Error 1
```

suborb, 2018-06-07:
It was just including the wrong filename - I renamed it several times before deciding on what it is now.

closed 2018-06-07 by suborb

Fabrizio-Caruso, 2018-06-07:
Is it fixed with the latest snapshots?

suborb, 2018-06-07:
I’ve just committed the change. So 20180608 will have it.

commit bc8d4a70 (2018-06-07): Fix #803: Include generic console in einstein for the variables
Fix #804: Get the name of the interrupt file right for sg1000