z88dk/z88dk #805: [m5] Latest snapshot: out of memory error at compile time

Fabrizio-Caruso, 2018-06-07:
Everything's ok with the snapshot of the 29th May.

On the more recent snapshots, I get:
```
rom: Embedded binary address range [8192,26565] exceeds ROM address range [8192,24575]
Building application code failed
make: *** [Makefile:1775: m5] Error 1
```

suborb, 2018-06-07:
I switched on the generic console by default in the m5 port, compile with -pragma-redirect:fputc_cons=fputc_cons_native and you'll get your memory back.

Fabrizio-Caruso, 2018-06-07:
Adding the native console breaks the compilation (I am trying to use ansi)

/cygdrive/c/z88dk/bin/zcc.exe +m5 \
-pragma-redirect:fputc_cons=fputc_cons_native \
-lm -create-app -Cz--audio \
-D__M5__ \
-clib=ansi -pragma-define:ansicolumns=32 \
-DFULL_GAME -DLESS_TEXT -DNO_SLEEP -DNO_WAIT \
-DBETWEEN_LEVEL -DEND_SCREEN \
-obuild/FULL_m5.bin \
src/end_screen.c \
src/horizontal_missile.c src/rocket.c src/item.c \
src/display_macros.c  src/enemy.c src/invincible_enemy.c \
src/level.c src/character.c \
src/text.c src/missile.c src/strategy.c src/input_macros.c \
src/main.c
Error at file 'c:/Z88DK/lib/../lib/crt/classic/crt_runtime_selection.asm' line 332: symbol 'DEFINED_fputc_cons' already defined
Error at file 'c:/Z88DK/lib/../lib/crt/classic/crt_runtime_selection.asm' line 334: symbol 'fputc_cons' already defined
2 errors occurred during assembly
Errors in source file c:\Z88DK\lib\config\..\..\\lib\target\m5\classic\m5_crt0.asm:
Error at file 'c:/Z88DK/lib/../lib/crt/classic/crt_runtime_selection.asm' line 332: symbol 'DEFINED_fputc_cons' already defined
                   ^ ----       defc DEFINED_fputc_cons = 1
Error at file 'c:/Z88DK/lib/../lib/crt/classic/crt_runtime_selection.asm' line 334: symbol 'fputc_cons' already defined
                   ^ ----       defc fputc_cons = fputc_cons_ansi
make: *** [Makefile:1776: m5] Error 1

suborb, 2018-06-07:
Oh sorry, that's my not thinking about you using the ansi library. Okay, that's an acceptable break there since you've got two overrides in play!

Can you compile with -m and provide a memory map with the two different versions? Something has got somewhat larger than expected/some extra stuff is being dragged in now.

Fabrizio-Caruso, 2018-06-07:
[maps.zip](https://github.com/z88dk/z88dk/files/2082255/maps.zip)

suborb, 2018-06-07:
Ah, okay, that makes sense, it's now trying to produce a ROM cartridge and not a tape file and your program is too big for M5 cartridges without a mapper.

Try compiling with -subtype=tape which will revert the behaviour.

suborb, 2018-06-07:
There's some notes on the wiki: https://github.com/z88dk/z88dk/wiki/Platform---Sord-M5 about the recent M5 work.

Fabrizio-Caruso, 2018-06-07:
THANKS! Problem solved!

closed 2018-06-07 by Fabrizio-Caruso