z88dk/z88dk #806: [cross chase][integration][test][feature] Cross Chase possible integration

Fabrizio-Caruso, 2018-06-07:
I am working on making Cross Chase as lean as possible by both removing unused code and files and by moving all devkit dependent code into a separate directory that can be excluded by selective checkout.
The source structure for devkit-dependent oprtion of my code will look like:
src/z88dk [z88dk related files only here]
src/cc65 [cc64 related files only here]
src/cmoc [cmoc related files only here]
src/wincmoc [wincmoc related files only here]
(I plan to add support to ack, small-c-85, gcc for ti and gcc in the future)

So, @suborb, can easily select a Z88DK only version by just avoiding 3 directories: src/cc65, src/cmoc, src/wincmoc (where only src/cc65 has many code and data files).

make z88dk_targets builds all z88dk targets (currently about 80 builds but only 40 targets because I support different memory and video configurations for many targets).

make z88dk_targets takes A LONG TIME 
Therefore to provide a faster test maybe excluding full versions and optimizations on targets build with zsdcc.
I may create provide some "make z88dk_test" to cover each target at least once.
@suborb, let me know what you think about it.

suborb, 2018-06-08:
That sounds good, we don't compile zsdcc regularly so any auto build should be just with sccz80 which should speed things up a bit.

I think I'll set this up to run on the nightly machines rather than through travis. We can then provide downloads as well if you like!

Fabrizio-Caruso, 2018-06-09:
@suborb , do you know how to do a selective checkout to avoid one CC65-related directory? I have read it is possible.

@suborb, that is nice of you to provide a download but I will keep on develop my project on my GitHub page and I would like to also have a link to the non-stable current development.

@suborb I will then write a "make test" that 
- only compiles with sccz80
- with zero optimization 
- only the TINY version  of my game with no extra features
This will basically tell us if the compilation of my sccz80 targets (i.e., all my targets except a few) is broken.
Since my code inside Z88DK is an almost frozen release (which we I will update from time to time), this test will provide a significant non-regression: I cover 40 targets with all sorts of output: native, low-level, sprites, ANSI/VT100, VT52, etc.

Fabrizio-Caruso, 2018-06-09:
@suborb 
I have now written a sccz80 testsuite to see if sccz80 can build my game.
It takes less than 3 minutes on a fast laptop (Core I7). I can still make it slightly faster.

I am not testing my game with these tests (I use just an unplayable super tiny version of my game). 

I am testing sccz80:
- all diverse compilations options used in CROSS CHASE (40 z88dk targets)
- sprites
- putc4x6
- native consoles
- VT100/ANSI
- VT52
- some target specific apis (e.g., cpcrslib)
- some low level routine (vpoke, etc..)
- in line assembly
and more

(My game can be unit-tested on the PC because I can now build a PC version with the VERY SAME game code)

Fabrizio-Caruso, 2018-06-09:
@suborb I was thinking of not running any tool during the compilation tests.
After all, these tests are just compilation tests and not functional tests.
The tool can be run as post-processing.

I also wonder whether you "can" include these third party tools. 
I suppose they are all free and open source.

The game produced by the compilation test is not enjoyable but it can be run. Some important game parts are missing: the enemies will not chase the player. So the game is trivial.

I don't understand the problem with the Galaksija. Why do you get a -final option?

The real functional game tests will be run in native PC mode.

Fabrizio-Caruso, 2018-06-09:
@suborb as far as the tool and the cfg directories are concerned, I was also planning to separate Z88DK from the other devkits.
I would like to have structure similar to the src directory:
tools/z88dk
tools/cc65
tools/gcc
...
cfg/z88dk
cfg/cc65
...
The cfg has pragma for Z88DK and linker configs for CC65, which are small txt files that take a few bytes.
On the other hand the tools directory may be too big for full inclusion in Z88DK.

So, if I separate tools, you could do at least selective checkout to skip:
src/cc65
tools/cc65
The rest non-Z88DK directories do not take much memory and some may be desired (PC target).

Fabrizio-Caruso, 2018-06-09:
@suborb I have just fixed a bug in sccz80_test and separate tools and cfg content into dev-kit specific directories.

Fabrizio-Caruso, 2018-06-10:
@suborb 
I have just seen a major change in zsdcc probably coming soon.
I have created a fast and lean little test also for zsdcc: "make zsdcc_test".
I only test the 8 targets that I need to build with zsdcc for my game 
for extra speed and to fit the game in less memory.

So you can now tests with:
make sccz80_test (about 40 targets)
make zsdcc_test (8 targets)
make z88dk_test (about 50 targets)

Fabrizio-Caruso, 2018-06-12:
@suborb, I will soon create a branch in my repository for the version that can be integrated into Z88DK.
I will NOT delete any CC65 (or any other devkit dependent directories). It is up to you to decide what to do with them (selective checkout? take all my project? take all my project but delete them...).
I am now working on a general clean-up and on adding some features I would like to see in the integrated version.

Fabrizio-Caruso, 2018-06-19:
Still re-factoring and testing my code... I want to do the big changes and tests before the integration.

Fabrizio-Caruso, 2018-07-05:
I am removing all possible code duplication among different dev-kits and targets. Now, even the display code is shared among all targets from CC65, CMOC, Z88DK, GCC+ncurses, etc.
The goal is zero code duplication. I am close to it.

Fabrizio-Caruso, 2018-07-29:
@suborb  Hi! 
I am ready to create an integration branch. I have significantly reduced any code duplication across different targets and different devkits.
What is left for me, is just some tests. 

Even the portion of the code about user-defined graphics is shared by CC65 and Z88DK.
With the only exception of the tools directory, where some tools are required specifically for some CC65 targets, (almost) all remaining code is shared. 

The only target-specific CC65 code is just:
- some very small header files with display configurations
- very small sound code
- some very small display initialization code 

So, now, you could simply take the entire repository and just remove the unnecessary tools. It is no longer worthwhile to do a selective check-out in my opinion. Nevertheless, I have kept the sound and display initialization code in separate directories for the CC65, Z88DK and other cases. So you can still remove the residual CC65-related code.

Fabrizio-Caruso, 2018-08-04:
@suborb 
I have created the branch: Z88DK_integration at
https://github.com/Fabrizio-Caruso/CROSS-CHASE/tree/Z88DK_integration

It may need some final testing but it is feature-complete for the purpose of initial integration.

Fabrizio-Caruso, 2018-08-10:
@suborb 
the Z88DK_integration branch above looks OK for integration.
Could you please let me know when you start integrating it into Z88DK?

suborb, 2018-08-14:
Sorry @Fabrizio-Caruso completely missed this.

My original plan was just to pull that branch nightly and build cross chase using the latest nightly of z88dk. 

When I get a chance I'll take a look and see if that's possible - I may well contribute some tweaks back to de-windows your build.

Fabrizio-Caruso, 2018-08-14:
No problem. I am using the extra time to add graphics to some targets by redefining some characters.
Cross Chase should not be a burden. I have optimized the code in all possible ways. I have reduced code duplication. Cross Chase should provide an additional way to test the devkit.

Fabrizio-Caruso, 2018-12-15:
@suborb By running my tests I have realized that tiki100 and probably abc80 are now broken.
I have opened issues for them.
We can improve stability if we integrate some automatic compilation tests such as the ones I can provide.
Is it still OK for you?
I can finalize them and ley you know? Ok?

suborb, 2018-12-16:
There's a few dependencies on Windows binaries that are blocking that at the moment.

I'll also need to introduce a few tweaks into the makefile to quieten it down a bit:

- Remove any -v options
- Introduce a $(Q) variable which will be prefixed to any command - this way I can make Q=@ and the command lines aren't printed
- Replace rm with $(RM) etc