z88dk/z88dk #809: zsdcc: update to upstream sdcc 10423

aralbrec, 2018-06-08:
After a bout of manual patching, I've successfully built an update to the current sdcc for windows vs2015.

A patch can be found here:
https://drive.google.com/file/d/1imOy_Kr-dLfy1-MRALQPxmDgnzWezhj9/view?usp=sharing

There are problems in some compiles which I still have to look at and I still have to go back and make sure all the proper functions have been patched.

spth, 2018-06-13:
To ease maintenance, and to increase the chance of some of the changes being merged upstream, I suggest to remove all cosmetic change (such as the indentation changes in src/z80/peep.c) from the patch.

Philipp

Fabrizio-Caruso, 2018-06-13:
I hope my game would still work and compile

feilipu, 2019-02-11:
Obsolete issue now.

closed 2019-02-11 by suborb