z88dk/z88dk #813: z80asm: need convenient way to generate unique (local) labels

aralbrec, 2018-06-12:
There is a current solution for doing this involving m4 which means the .asm file must be changed to .asm.m4 and relatively clumsy m4 syntax must be used.

Moving to an m4 file has some disadvantages in that errors are reported in .asm files instead of .asm.m4 files so that line numbers can be widely different.  It also means more care has to be taken not to conflict with m4 syntax like watching for stray apostrophes, backticks, etc.  m4 makes it possible to do otherwise impossible things but for something trivial like local label generation, it's an unnecessary annoyance.

The current solution using m4 involves two macros `Z88DK_ULBL` and `Z88DK_CLBL` for generating unique labels and current labels respectively.  They are used as follows:

`Z88DK_ULBL` generates a new unique label "__uniq_xxxx" where xxxx is a 4-digit decimal number.
`Z88DK_ULBL(prefix)` generates a new unique label "prefix_xxxx"

In both these cases the decimal number increases from 0000 by one separately.  So the first unique label will be "__uniq_0000" and the first prefix label (where prefix can be any string) will be "prefix_0000" as well - the sequences are separate.

`Z88DK_CLBL` generates the last unique label created "__uniq_xxxx"
`Z88DK_CLBL(-1)` generates the unique label created before the last one ie "__uniq_xxxx-1"

The `-1` displacement can be any displacement negative or positive so that you can refer to previously made labels or to labels made later on.

The prefix counterpart:

`Z88DK_CLBL(,prefix)` generates the last unique label created "prefix_xxxx"
`Z88DK_CLBL(-1,prefix)` generates the last unique label created before the last one ie "prefix_xxxx-1"

An example that should clarify what these things do:

```
include(`z88dk.m4')

Z88DK_ULBL:
Z88DK_ULBL:
Z88DK_CLBL:
Z88DK_CLBL:
Z88DK_CLBL(-1):
Z88DK_ULBL:
Z88DK_ULBL(prefix):
Z88DK_ULBL(prefix):
Z88DK_CLBL:
Z88DK_CLBL(-1):
Z88DK_CLBL(,prefix):
Z88DK_CLBL(5,prefix):
Z88DK_ULBL(prefix):
```

becomes:

```
__uniq_0000:
__uniq_0001:
__uniq_0001:
__uniq_0001:
__uniq_0000:
__uniq_0002:
prefix_0000:
prefix_0001:
__uniq_0002:
__uniq_0001:
prefix_0001:
prefix_0006:
prefix_0002:
```

A macro using unique labels:

```
define(`TILEPRINT',
`
IF offsey & 1
   ld (Z88DK_ULBL(_tp)+1),a
   CELLPRINT($f901)
   CELLPRIN1
   CELLPRINT($f901)
   CELLPRIN2
Z88DK_CLBL(,_tp):
   ld a,0
ELSE
   CELLPRINT($f901)
   CELLPRINT($f91f)
   CELLPRINT($f901)
   CELLPRINT($f8e1)
ENDIF
')
```

aralbrec, 2018-06-12:
I'd like to propose adding the same label generation in some sort of label syntax suitable for z80asm and that doesn't interfere with m4.  I'm not sure what that syntax should be yet.

pauloscustodio, 2018-06-12:
Z80MR from CP/M has the syntax:
```
local aaa
....
aaa
```
where each ```local aaa``` will generate a new unique label with prefix ```aaa```.

Your example would be:
```
define(`TILEPRINT',
`
LOCAL _tp
IF offsey & 1
   ld (_tp+1),a
   CELLPRINT($f901)
   CELLPRIN1
   CELLPRINT($f901)
   CELLPRIN2
_tp:
   ld a,0
ELSE
   CELLPRINT($f901)
   CELLPRINT($f91f)
   CELLPRINT($f901)
   CELLPRINT($f8e1)
ENDIF
')
```
There is no syntax in Z80MR to refer to previous or future labels; maybe something like ```LOCAL(aaa,offset)```?

aralbrec, 2018-06-12:
Maybe:

```
local aaa(-2)
local aaa
local aaa(+5), bbb, ccc(-1)
```

Does Z80MR have a way to turn off local substitution?

```
local loop

loop:
   djnz loop

unlocal loop   ;??
endlocal loop   ;??
localend loop   ;??

;;; here I want to use "loop" not a unqiue label with prefix "loop"

loop:
   jr loop

;;; here I turn it on again

local loop

loop:    ;; unique label next in the sequence so different from first one
   djnz loop
```

pauloscustodio, 2018-06-12:
In Z80MR, the local label scope is limited to the current macro expansion, i.e. between MACRO and ENDM.
Maybe we can use LOCAL / ENDLOCAL to limit the scope.

I do not see the use case for the label(offset) usage? Isn't it cleaner and easier to understand to define a fixed name as the auto-label in a defc?

aralbrec, 2018-06-13:
> In Z80MR, the local label scope is limited to the current macro expansion, i.e. between MACRO and ENDM.
> Maybe we can use LOCAL / ENDLOCAL to limit the scope.

I think you're planning to use MACRO / ENDM for macros so maybe LOCAL / ENDL is congruent?

aralbrec, 2018-06-13:
> I do not see the use case for the label(offset) usage? Isn't it cleaner and easier to understand to define a fixed name as the auto-label in a defc?

It would be for macros mainly, allowing them to refer to previous instances of the same macro expansion.  I don't think you would see it in typical use for code labels.

aralbrec, 2018-07-26:
Adding a specific case to this, it would be nice to be able to define an implied symbol for the current assemble address, eg something like:

```
jr $
```

with meaning:

```
__internal_symbol:

   jr __internal symbol
```

for compatibility with non-linking assemblers and for convenience.

This is different from `ASMPC` which is a byte displacement into the current section in the current file.

pauloscustodio, 2018-07-26:
> jr $

Easy thinks first. This is done.