z88dk/z88dk #814: z80asm macros

maziac, 2018-06-14:
Hi, 
sorry to put this here, but I haven't found any accurate place.

I'm trying to use macros with the z80asm Assembler but I can't find any documentation
about it.
How can I put macros in the assembler code (not in C but in pure assembler files) ?
What is the syntax?

pauloscustodio, 2018-06-14:
z80asm does not have macro support. The feature is planned, but not yet done.
The current solution is to run the z80asm source through m4 before assembly. 

I will leave this topic open, to close when macros are implemented.

pauloscustodio, 2018-06-14:
There is a Perl script that you can run with ```perl -S asmpp.pl file.asm``` that implements the macro syntax used by the CP/M assembler Z80MR and calls z80asm:
```
name MACRO #param1,#param2
          LOCAL #local_label
#local_label:
          ....
          ENDM

          name 12, 34
```

aralbrec, 2018-06-14:
M4 is integrated as a macro processing language.

Basic information:
https://en.wikipedia.org/wiki/M4_(computer_language)

Introduction:
http://mbreen.com/m4.html

aralbrec, 2018-06-14:
If a file ends in ".m4" then zcc will pass it through m4 before the next stage.

"foo.asm.m4" will pass through m4 to generate "foo.asm" (in the same directory!) which will then be passed to the assembler as usual.

"foo.c.m4" will pass through m4 to generate "foo.c" (in the same directory) which will then be passed to the c compiler as usual.

A couple of special filetypes ".inc.m4" and ".h.m4" will have m4 applied to result in ".inc" and ".h" respectively but processing stops there.  These can then be used as includes into asm or c.

pauloscustodio, 2021-12-28:
z80asm is now a macro assembler!

Implemented MACRO/ENDM, REPT/ENDR, REPTC/ENDR, REPTI/ENDR, EXITM, LOCAL, DEFL, #define, #undef

Documentation at https://github.com/z88dk/z88dk/wiki/Tool---z80asm---preprocessor

closed 2021-12-28 by pauloscustodio