z88dk/z88dk #817: [zsdcc][zx81][--max-allocs...] zx81 produces right code only with --max-allocs-per-node200000

Fabrizio-Caruso, 2018-06-17:
Somthing weird happens with the zsdcc and the zx81 target.
If I compile it without --max-allocs-per-node200000 I get a malfunctiong binary (corrupted game with player moving outside the main area).

Then I suppose that --max-allocs-pernode200000 must be somehow replacing the wrong bits if code with the right ones. Maybe this info can help you find the bug.

aralbrec, 2018-06-17:
We fix some code generation bugs in post processing when the peephole optimization is SO2 or SO3.  I can see now that the classic is mistakenly using a default value of SO0 (no peephole optimization) when it should be using SO2 as default.  Try the compile at SO2 or SO3 level and the bugs should go away at all max-allocs settings.

SO3 is an aggressive set that can sometimes make a big difference in code size & performance.  Adding "--opt-code-size" can further reduce code size.

aralbrec, 2018-06-17:
There may be a bug we missed.  Do you know which function(s) it may be located in?

Fabrizio-Caruso, 2018-06-17:
I will try to isolate the issue. What I see in my game looks like an issue with the player's position. It looks like the y coordinate is set to zero. I will need to do more experiments to isolate the issue.

Fabrizio-Caruso, 2018-06-17:
@aralbrec  by the way I know the --opt-code-size option but I cannot use it with the zx81 target because it breaks the binary. I am reporting these problems on the zx81 target. The other targets seem ok more or less.

Fabrizio-Caruso, 2018-06-17:
@aralbrec --opt-code-size deserves an issue of its own: it breaks multiple targets if used alone.

Fabrizio-Caruso, 2018-06-17:
@aralbrec 
-SO3 -> KO
--opt-code-size -> KO
-SO3 --opt-code-size --max-alloc -> KO
--opt-code-size --max-alloc -> KO
--opt-code-size -SO3 -> KO
but
--max-alloc -> OK
-SO3 --max-alloc -> OK

aralbrec, 2018-06-19:
The low max-allocs-per-node is generating buggy code so this will mean buggy code is processed by SO3 and opt-code-size.

I'll have to see if I can spot the bad code generation.  If you know which file and which function(s) might be involved, let me know along with the compile line so I can reproduce it.

Fabrizio-Caruso, 2018-06-19:
I think it may not be easy to detect the function.
I have an "improbable candidate". If compile my game with the -DNO_BLINKING option, I exclude the blinking function and the zx81 seems to work:
```
void _blink_draw(unsigned char x, unsigned char y, Image * image, unsigned char *blinkCounter) 
{
	if(*blinkCounter) 
	{
		_draw(x,y,image);
		*blinkCounter=0;
	} 
	else 
	{
		_delete(x,y);
		*blinkCounter=1;
	}	
}
```
Where _draw and _delete may be functions or macros.
This function is placed in different files depending on the target.
For the zx81 target it is in:
src/cross_lib/z88dk/zx81
in the newest version of my game (that is going through a significant re-factoring).

Fabrizio-Caruso, 2018-06-21:
@aralbrec SO2 does not fix the compilation issue with my game.
Only -max-alloc without --opt-size produces the right binary and SO2/SO3 have no effect on the bug.

Fabrizio-Caruso, 2018-06-21:
@spth, it seems SDCC is produces wrong binaries without --max-alloc  or with --opt-size when used with my code.
I am not sure I can reproduce it with a simple example.
It looks like the innocent looking _blink_draw function in my code is somehow triggering the issue because if I disable that function, I don't see the issue.

Remark: the game works fine with sccz80, cc65, cmoc, wincmoc, gcc and other c compilers.

Fabrizio-Caruso, 2018-06-21:
@spth, it could just be a problem with the display routines and zsdcc and not just a zsdcc problem.
It looks like the vg5k target is not impacted: I can compile it with the blinking function above and without the --max-alloc... option. The game seems to work.
On the other hand the Spectravideo target does not work with zsdcc without --max-alloc...
I need to run more tests...

Fabrizio-Caruso, 2018-06-21:
@spth, on both the svi and zx81 target the presence of the blinking function is associated with the corrupted binary.
On the vz200 and vg5k I do not see this problem.

spth, 2018-06-21:
Just a guess: AFAIK z88dk uses sdcc with --reserve-regs-iy. There were a few bugs in upstream SDCC in code generation with --reserve-regs-iy that got fixed upstream half a year ago. Maybe you are running into one of these?
If yes, trying this: https://github.com/z88dk/z88dk/issues/809 might make sense.

Philipp

aralbrec, 2018-06-21:
We fix a number of those issues at SO2 level.  zsdcc is fairly stable now - there are many large programs being successfully compiled using it but that doesn't mean we caught all instances.

It's a bit strange that identical code runs fine on one target and not another so it can also be library related, possibly in how parameters are passed as that's different between sccz80 and zsdcc.

I'm being pulled in about 10 different directions but I'll see if I can spot a problem when I get a chance.

SO2 was made default in the classic lib because that's what it should be; I didn't expect it to fix this problem :)

Fabrizio-Caruso, 2018-09-03:
@aralbrec  has the commit in 
https://github.com/z88dk/z88dk/commit/6be150136365f128243155f8f40a87aaa0b30dbe
anything to do with this issue?

aralbrec, 2018-09-03:
No I do not think so.  This commit will affect "-SO3" and "-SO3 --opt-code-size" compiles

Fabrizio-Caruso, 2018-09-03:
@aralbrec  I have not decribed the issue clearly here. 
The problem I am having is exactly with -SO3 and -SO3 --opt-code-size.
My problem is somehow fixed if I add --max-allocs-per-node200000.
So, I need to test your fix. You may have solved this issue.

Fabrizio-Caruso, 2018-09-04:
@aralbrec 
I have all cases with zx81 + sdcc with all optimization combinations with today's (4th of Sept) snapshot.
zx81 + sdcc compiles a functional binary with my game if and only if the -max-allocs options is used:

KO cases:
- no optimization
- SO3
- opt-code
- SO3 -opt-code

OK cases:
- max-allocs
- max-allocs -opt-code
- max-allocs -SO3
- SO3 -max-allocs -opt-code

"
	$(Z88DK_PATH)$(MYZ88DK) +zx81 \
	$(ZSDCC_OPTS) \
	-compiler=sdcc \
	-vn \
	-D__ZX81__ -DFULL_GAME -DEND_SCREEN -DBETWEEN_LEVEL \
	-DCONIO -DNO_INIT_GRAPHICS \
	-lndos \
	-create-app -o  $(BUILD_PATH)/FULL_zx81_16k.prg \
	$(SOURCE_PATH)/end_screen.c \
	$(FULL_FILES)
"

Maybe I will open a zx81 specific issue. First, I need to figure out how other targets are behaving.

Fabrizio-Caruso, 2018-09-14:
@aralbrec The issue is still present. Maybe I am doing something stupid with the compilation options.
Should I compile with  --reserve-regs-iy ??

aralbrec, 2018-09-14:
Yes for zx81 you have to compile with --reserve-regs-iy and you must make sure you do not have any large local variables declared (which would only be the case in programs ported from big machines).

aralbrec, 2018-09-14:
I'm still working on another SO3 bug so there may be another fix today.

suborb, 2018-09-14:
Obviously --reserve-regs-iy has an issue in classic with #970 - I'll endeavour to fix that for the build tonight though.

Fabrizio-Caruso, 2019-05-19:
I now get a correct looking binary when I compile Cross Chase for the ZX81 target with just -compiler=sdcc and no optimization option. It seems that something fixed a long standing bug. I guess something was improved in zsdcc.

suborb, 2019-05-19:
We'll be upgrading sdcc again at some point soon, so I'll be prepared for you reopening this issue!

closed 2019-05-19 by suborb