z88dk/z88dk #819: [zsdcc][zx81][vz200] --opt-code-size breaks the binaries

Fabrizio-Caruso, 2018-06-17:
On multiple targets I can see that if I compile with --opt-code-size (with ot without --max-allocs...) I get a malfunctioning binary. Maybe related to the issue about the positive effect of --max-allocs.

What I can see, when compiling the full version of my game, is:
--opt-code-size breaks the binary
--max-alloc... fixes the binary if --opt-code-size
-SO3 does not have any effect

So for example it looks like (only tested with zx81, vz200, svi and not all cases with all targets)
-SO3 -> KO
--opt-code-size -> KO
-SO3 --max-alloc -> KO
--max-alloc -> KO
-SO3 --opt-code-size --max-alloc -> KO
--opt-code-size --max-alloc -> KO
but
--max-alloc -> OK
-SO3 --max-alloc -> OK

aralbrec, 2018-06-19:
Yes it might all come down to buggy code from the low max-allocs setting.  opt-code-size could be transforming the buggy code so it's not fixed in combination with max-allocs.

spth, 2018-06-21:
Well, SDCC should never generate broken code, no matter how low --max-allocs-per-node is set. It might generate slower, bigger code that uses more memory, but the code should still work.
So if you can create a small reproducer that shows the problem also with current SDCC, please file a bug report with upstream SDCC.

Philipp

feilipu, 2019-07-05:
Is this still the case?
Please check with r11311.

zx70, 2019-07-08:
Fabrizio told me he's checking the new sdcc version.    reviving the topic

Fabrizio-Caruso, 2019-07-08:
@zx70 This issue seems solved now! :-)

closed 2019-07-08 by Fabrizio-Caruso