z88dk/z88dk #821: [question/bug] Meaning of "stdlib.h:295: syntax error: token -> '__z88dk_fastcall'"

Fabrizio-Caruso, 2018-06-21:
I get a failed compilation failure with no much information:
```
...
type "C:\cygwin\tmp\zcc13CE4.tc:/Z88DK/lib/config/../..//include/stdlib.h:295: syntax error: token -> '__z88dk_fastcall' ; column 27
...
make: *** [Makefile:517: vz200_8k] Error 1
```

Fabrizio-Caruso, 2018-06-22:
$(Z88DK_PATH)$(MYZ88DK) +vz -v \
	-pragma-include:$(CFG_PATH)/z88dk/zpragma.inc \
	-compiler=sdcc \
	-opt-code-size \
	-lndos \
	-create-app -o $(BUILD_PATH)/TINY_vz200_8k.vz \
...

I found out that it was a macro related issue.

aralbrec, 2018-06-22:
There's no error on that line but there are some expanding macros there.
Is it working for you now with the macro issue fixed or is it still behaving improperly?

Fabrizio-Caruso, 2018-06-22:
@aralbrec  I have fixed the my specific issue.

I have a question or feature request for the future macro-related bug investigations:
is there a way to just run the pre-processor and see what the code looks like after all macros are applied?

aralbrec, 2018-06-22:
You can add "-E" to the zcc line and it will stop after the preprocessor step.

closed 2018-06-22 by Fabrizio-Caruso