z88dk/z88dk #823: [trs80] getk does not detect continued key-presses

Fabrizio-Caruso, 2018-06-26:
getk() does not detect continued key-presses.

Instead, I use:
```
		#define POKE(addr,val)     (*(unsigned char*) (addr) = (val))
		#define PEEK(addr)         (*(unsigned char*) (addr))
		unsigned char ijkl = PEEK(0x3802);
		if(ijkl==2)
			return 'I';
		else if(ijkl==4)
			return 'J';
		else if(ijkl==8)
			return 'K'; 
		else if(ijkl==16)
			return 'L';
		if(PEEK(0x3840)==128)
			return ' ';
		else
			return '\0';
```

suborb, 2018-06-26:
@Fabrizio-Caruso your timing is impeccable!

Just this morning I committed an inkey() driver for the TRS80 and EG2000, enable it with `-pragma-redirect:fgetc_cons=fgetc_cons_inkey`

It reads the memory in the same way as your snippet above.

I've also just added drivers for the Sorcerer, Aquarius and VG5k

closed 2018-06-26 by suborb

Fabrizio-Caruso, 2018-06-26:
@suborb, does this fix getk() as well?
As you know I am trying to use the smallest possible number of different input routines.
My goal is to write a universal cross-dev-kit tool-chain (I am almost there and my game is a proof of concept).

suborb, 2018-06-26:
Good point, you'll need `-pragma-redirect:getk=getk_inkey` to switch over getk.

I'm focussing on those platforms that have issues with the firmware implementations - I suspect Nascom might be next on my list.

Fabrizio-Caruso, 2018-06-26:
@suborb, as far as Nascom is concerned @zx70 worked on it and got something useful, yet not perfect.

As you know, I am testing nearly all targets from all dev-kits.
For Z88DK, I get some usable on most targets now.

The real trouble makers now are TI82, TI83, TI85 (as I have already reported in other issues).

Other minor issues exist in other targets such as SVI, wich has slow response time (it must be using the kernel routines).

I would like to have some uniform routines to have input functions that are responsive enough for action games. The kernel routines not always provide this (e.g., you get no continued key-presses or delayed detection of continued presses).

Fabrizio-Caruso, 2018-06-26:
@suborb THANKS A LOT for your huge efford!

The worst input is TI85's one (issue #780 ) followed by TI82 and TI83 (issue #783 ).