z88dk/z88dk #824: [getk][inkey][spectrum][ansi] -pragma-redirect:getk=getk_inkey  won't compile

Fabrizio-Caruso, 2018-06-26:
I would like to use getk on the spectrum target as I do with the other 40 targets but default getk is slow. 
in_Inkey() is very responsive.
If I try to compile with:
-pragma-redirect:getk=getk_inkey:
```
/cygdrive/c/z88dk/bin/zcc.exe +zx -O3 -clib=ansi -pragma-redirect:getk=getk_inkey -v  \
-pragma-redirect:ansifont=_udg -pragma-define:ansifont_is_packed=0 -pragma-define:ansicolumns=32 \
-DFULL_GAME -DREDEFINED_CHARS -DSOUNDS -DCLIB_ANSI -DEND_SCREEN -DBETWEEN_LEVEL -D__SPECTRUM__ \
-DALT_MOVE \
-lndos -create-app \
-o ../build/FULL_spectrum_48k.prg \
```
I get these errors:
```
...
z80asm -I"c:/z88dk/\lib\target\zx\def" -D__SCCZ80  -s  --cpu=z80 -I"c:/z88dk//lib" -I"c:/z88dk/\lib\target\zx\classic"  "D:\personal\cygwin64\tmp\zcc4B4120.asm"

z80asm --cpu=z80 -b -d  -o"../build/FULL_spectrum_48k.prg" -m -s -L.   -D__SCCZ80 -L"c:/z88dk//lib/clibs" -I"c:/z88dk//lib" -indos -izx_clib -i"z80_crt0"     "@D:\personal\cygwin64\tmp\zcc4B4121.lst"
Errors in source file c:/z88dk/\lib\target\zx\classic\spec_crt0.asm:
Error at file 'zcc_opt.def' line 6: symbol 'getk_inkey' not defined
                   ^ ----       defc getk = getk_inkey
make: *** [Makefile:1011: spectrum_48k] Error 1
```

suborb, 2018-06-26:
The redirector file wasn't included in the library, it is now so it'll be there tomorrow.

I also fixed the zx81 as well since that had the same problem.

closed 2018-06-26 by suborb

suborb, 2018-06-27:
Yup. They are exported for sdcc so should work.

Fabrizio-Caruso, 2018-06-27:
Without ansi, can I use getk with the spectrum target and clib=sdcc_iy?

I get:
```
/cygdrive/c/z88dk/bin/zcc.exe +zx --opt-code-size  -startup=1 -zorg=24055 \
-pragma-redirect:getk=getk_inkey \
-pragma-include:chase/../../cfg//z88dk/zpragma.inc -clib=sdcc_iy \
-DNO_SLEEP -DLESS_TEXT -DTINY_GAME -v  -D__SPECTRUM__ \
-DEXT_GRAPHICS \
-create-app -o ../build/TINY_spectrum_16k.prg \
cross_lib/display/z88dk/spectrum/spectrum_graphics.c \
cross_lib/display/display_macros.c cross_lib/input/input_macros.c cross_lib/text/text_macros.c chase/move_player.c chase/ghost.c  chase/level.c chase/character.c chase/game_text.c chase/main.c chase/strategy.c
cross_lib/input/input_macros.c:155: warning 112: function 'getk' implicit declaration
cross_lib/input/input_macros.c:155: warning 84: 'auto' variable 'getk' may be used before initialization
cross_lib/input/input_macros.c:155: warning 84: 'auto' variable 'getk' may be used before initialization
zpragma < "chase/../../cfg//z88dk/zpragma.inc" > nul
...
Errors in source file cross_lib/input/input_macros.c:
Error at file 'cross_lib/input/input_macros.c' line 831: symbol '_getk' not defined
                   ^ ---- (null)make: *** [Makefile:1000: spectrum_16k] Error 1
```

suborb, 2018-06-27:
-clib=sdcc_iy is the new library which doesn’t have getk in it. 

To use classic with sdcc you need to use the -compiler=sdcc option