z88dk/z88dk #825: [spectrum][kbhit] -pragma-redirect:getk=getk_inkey breaks kbhit and getk.

Fabrizio-Caruso, 2018-06-27:
-pragma-redirect:getk=getk_inkey  breaks the kbhit:

If I use -pragma-redirect:getk=getk_inkey no input is eve detected.

suborb, 2018-06-27:
Have you got a test case for this? - I've just tried out the following:

    #include <stdio.h>
    int main() {
        while ( 1 ) {
            if ( kbhit() ) {
                int c = getch();

                printf("%c",c);
            }
        }
    }

And it prints the pressed key as expected

Fabrizio-Caruso, 2018-06-27:
So far, I have only used my game. I will try to isolate the issue.
In my game I am not using getch(). I use getk() instead, which should also work.

suborb, 2018-06-27:
Yes, replacing the getch() with a getk() and a zero check works as expected.

Fabrizio-Caruso, 2018-06-27:
I need kbhit to check if there has been any input. All other getk implementations work fine if preceded with a kbhit check. It seems that kbhit is preventing getk to work.

Fabrizio-Caruso, 2018-06-27:
You are right. kbhit + getch should be the same as kbhit + getk. I need to try this...

Fabrizio-Caruso, 2018-06-27:
It does not work. I get no input.
I am doing:
```
if(kbhit())
{
         movePlayerByKeyboard(getch()); 
}
```
kbhit() is always false or getch() is detecting the wrong keys. No kets apparently.

I am compiling with:
```
	$(Z88DK_PATH)$(MYZ88DK) +trs80 -lndos \
	-pragma-redirect:getk=getk_inkey \
	-lm -create-app \
```

Fabrizio-Caruso, 2018-06-27:
Moreover kbhit followed by getk should also work but it does not.
Is kbhit consuming the input? It should not consume it. It should just detect it.

suborb, 2018-06-27:
The way the inkey drivers work is by instantaneously reading the keyboard, so what is pressed at that instant - there's no concept of a buffer or anything like that.

My example showed that kbhit() followed by getch() is working correctly....

One thing that might be different is that the inkey driver supports the shift and "control" key modifiers. With no modifiers pressed, it will return the **lower case** character. This is possibly different to the firmware drivers (which may be stuck in upper case mode).

Fabrizio-Caruso, 2018-06-27:
From what I can see in my game is that kbhit() never detects any key, even a wrong key:
With
```
if(kbhit())
{
	movePlayerByKeyboard(_MOVE_LEFT); 		    
}
```
I should have my player move to the left whatever key is pressed.

If I remove kbit() I see the player move (which means _MOVE_LEFT is correct).

getk() is also not behaving correctly:
movePlayerByKeyoard(getk())
never detects any key.

I am testing my game with trs80gp emulator under Windows.

Fabrizio-Caruso, 2018-06-27:
@suborb  something fishy is going on with kbhit, if I copy the very same code inside my routine, kbhit does not work. It does work in isolation.
The following code works
```
int main() {
    while ( 1 ) {
        if ( kbhit() ) {
            printf("%c",getch());
        }
    }
}
```
but if I use it inside my routine, then kbhit() is always false:
```
	void MOVE_PLAYER(void) 
	{ 
	
        if ( kbhit() ) {
            printf("%c",getch());
        }	
```
I am compiling my game and the text example with the very same options:
```
	$(Z88DK_PATH)$(MYZ88DK) +trs80 -lndos \
	-lm -create-app \
	-pragma-redirect:getk=getk_inkey \
	-vn \
	...
```

suborb, 2018-07-03:
Ok, I've finally figured this one out via the forum.

When using inkey() kbhit returns a keypress once every 256 calls. When you're busy looping this isn't a problem. But when you're printing on screen it is an issue.

I need to adjust the key debounce parameters to get it more responsive.

suborb, 2018-07-03:
@Fabrizio-Caruso I've just merged this, hopefully it will sort out your issues

closed 2018-07-03 by suborb