z88dk/z88dk #833: (classic) Add Samsung SPC-1000 support

suborb, 2018-07-02:
* Z80 @ 4Mhz
* MC6847
* AY-3-8910 @ 4MHz
* 64k RAM
* 32k ROM (Hudson Basic)

Bare bones:

- [x] Appmake support for +spc1000
- [x] Native console output
- [x] Inkey driver

Better:

- [ ] Generic console
- [ ] PSG support

Best:

- [ ] Graphics

suborb, 2018-07-05:
Merged as #834

closed 2018-07-05 by suborb

zx70, 2018-07-17:
I'v been looked for the Joystick code in the ROM (PSG instructions and BASIC keywords) but I wasn't able to recognize it (if it ever exists).
This code portion comes from the TS2068 technical manual (also not yet implemented in z88dk):

; joystick (player number = 1 or 2)
;Note that PSG Register 7, Bit 5 is assumed to be zero, enabling I/O Port A for input. If you
;ever use I/O Port A for output (R7,B6=1), you will want to clear Bit 6 prior to any input operation

LD A,OEH         ;Load A = 14
OUT A,(0F5H)    ;Address the joystick port
LD B,playerno
LD C,0F6H         ;Data Port address to C
IN A,(C)            ;Joystick data to A
CPL                  ;Complement to High Active
AND 8FH          ;Get significant bits