z88dk/z88dk #836: (classic) Add Sanyo PHC-25 support

suborb, 2018-07-04:
_This is a placeholder, still seeking ROMs and need to reverse engineer the tape format_

* Z80 @ 4MHz
* MC6847 + 6k VRAM
* 16k RAM
* 24k ROM + 4KB chargen
* Keyboard 0x80 - 0x88 (https://github.com/mamedev/mame/blob/master/src/mame/drivers/phc25.cpp for keymap)
* AY on 0xc0 and 0xC1
* Two joysticks on AY port A and B

Links:

* Takeda emulator 
* http://www.geocities.jp/sanyo_phc_25/
* http://www.phc25.com

suborb, 2018-07-06:
@zx70 - the ROM is apparently a Microsoft BASIC derived, but basck is failing to identify anything in it.

zx70, 2018-07-06:
I think I tested it on the phc70 system disk and it worked,  I'll investigate

zx70, 2018-07-07:
ok, I  was thinking at the sony SMC-70 / SMC-777.    I began hunting material for the sanyo (e.g. the MBC200 and MBC 1200) but it was very preliminary and parsing the boot disk image with basck was useless..  I'll focus on that rom (

zx70, 2018-07-07:
Ok, it is not a microsoft BASIC.    As for the SC3000 I'm not able to classify it, it can also be an original work  :)
Basck fails here and there (e.g it can't detect the HuBASIC used on the mz2500 disks), but afaik ALL the MS basic answer '0' if you type "? a".
HuBASIC and others replicate the shortcuts, but I consider it a valid test to exclude Microsoft..

zx70, 2018-07-09:
Short update:  "GETVAR" is a wrong assumption, the code at $0020 seems to have to do with the recognition of the characters in a basic program.
'basck' is able to detect the MZ2500 BASIC as HuBasic, it failed because the data bits in an mz2500 disk image are inverted.

zx70, 2019-01-24:
So, most probably the Sanyo BASIC is not an MSBASIC or HuBASIC variant.
I chose a manual approach and spotted the entries for an extra simple library.
I suspect that different ROM versions could be troublesome for fputc_cons().  I'm not expecting problems with fgetc_cons(), but getk() will be as limited as for #884

suborb, 2019-01-24:
@zx70 cool. Though we're still getting files onto the machine - I don't really want us to have to the ABC80 hex thing here, but I suspect that might be what we're left with.

zx70, 2019-01-24:
For the record, the ABC80 hex thing must die sooner or later, I'm not nostalgic about it  :D

suborb, 2019-01-24:
I think screen memory might be the way to go - I think the BASIC uses \0 as an end marker rather than having the length of the line.

I have a mangled SPC-1000 font that I was using. Upper case works, but lower case is mangled somewhat. I'll try again to build a working font.rom.

suborb, 2019-01-24:
I was of course being an idiot. 031_04a.ic6 should be renamed FONT.ROM and then it's all good. wowroms has the ROMset.

suborb, 2019-01-24:
The MAP-1010 is very similar hardware. I think the keyboard reading is different so it needs an alternate BASIC.ROM, I'm still searching for that one.

zx70, 2019-01-24:
I miss that font rom.. can I have it ?

suborb, 2019-03-28:
@zx70 sorry, I missed this, do you still the rom or are you good? The site above has the working set.

zx70, 2019-03-28:
no worries, I should have all.   In any case I'm diverted at the moment  :)

zx70, 2022-10-24:
https://github.com/z88dk/techdocs/blob/master/targets/phc25/phc25rom.asm

zx70, 2023-03-24:
I just succeded in decoding the floppy disk formats of the Sanyo mbc-200 (MAME emulates it, but beware, a second floppy disk might not be detected easily, if in trouble try accessing drive F:, forcing CP/M to ask to change the disk helps).

I found definitions for "san1" here:
https://forum.vcfed.org/index.php?threads/sanyo-mbc-1250.15769/

I derived 'san2' from it, it looks obvious now but it hasn't been easy at all !

# SAN1 Sanyo MBC-1000, MBC-1150 - DSDD 48 tpi 5.25" - 256 x 16
diskdef san1
seclen 256
tracks 80
sectrk 16
blocksize 2048
maxdir 64
skew 3
boottrk 2
os 2.2
end

# Sanyo MBC-1200/1250 - DSDD 96 tpi 5.25" - 256 x 16
# Probably valid also for MBC-250/225/200
# 256 byte sector, 1-16, 1:1, 96 tpi
diskdef san2
  seclen 256
  tracks 160
  sectrk 16
  blocksize 4096
  maxdir 128
  skew 3
  boottrk 4
  os 2.2
end

Fabrizio-Caruso, 2025-05-31:
Hi everyone! Is this target usable? How?

Fabrizio-Caruso, 2025-05-31:
Basic program with data statements would make it useable for small programs I suppose.

barbeque, 2025-06-01:
It does seem relatively straightforward to generate a *.PHC tape file, as it appears to be a one-for-one byte stream. Here's a sample of `03-sanyo-phc25-basic2_11k8b.phc`:

[image]

A PC-6001-style `DATA` statement with an assembly loader that switches to stage 2 would probably be enough to load a large machine language program...

zx70, 2025-06-01:
I'm not sure I understood you suggestion, do you mean the BASIC DATA instruction of something different?

barbeque, 2025-06-02:
> I'm not sure I understood you suggestion, do you mean the BASIC DATA instruction or something different?

The BASIC `DATA` keyword. Inufuto, a Japanese author who does a lot of ports to different systems, has a target for the PHC-25 in his high(er)-level language, Cate. I haven't had a chance to look into it deeply, but a friend linked me to a good example for his recent game Mazy2.

Here's his assembly loader: https://github.com/inufuto/Cate_examples/blob/main/phc25/mazy2/Loader.asm

And the code that generates his BASIC one: https://github.com/inufuto/Cate_examples/blob/main/phc25/mazy2/Loader.csx

Here's the makefile that sticks those two together: https://github.com/inufuto/Cate_examples/blob/main/phc25/mazy2/makefile

suborb, 2025-06-17:
Current state:

* MC6847 library is mostly split out
* I can create a loadable PHC file with a BASIC bootstrap program
* We already had some ROM entry points + PSG functions

Todo:

* Mode switching for the MC6847
* Create a working loader
* Append the actual program to the phc file

barbeque, 2025-06-17:
Thank you very much for your hard work so far!

suborb, 2025-10-04:
If I remember I was struggling with reliable loading. It worked in one emulator but not in another.

I think I also ran out of patience with how slow the loading was so parked it for a while tbh.

Fabrizio-Caruso, 2025-10-06:
@suborb @zx70 
Some very recent homebrews for the Sanyo PHC-25 :
https://itch.io/jam/rrpufos-sanyo-phc25/entries

I share here some doc from the FB group (https://www.facebook.com/groups/1161811471271807/) that organized the game jam:
[phc25.full.txt](https://github.com/user-attachments/files/22716291/phc25.full.txt)

suborb, 2025-10-06:
Thanks, that's given me another avenue to try - maybe this:

>end of file is indicated by 0,0. 

isn't strictly true.

Fabrizio-Caruso, 2025-10-06:
Maybe the discord channel of the game jam can provide some extra info (in French though):
https://discord.com/channels/1373597736464547980

suborb, 2025-10-06:
I'm not too sure what else is actually working, but there's firmware keyboard reading and (at least) text mode --generic-console

Now I've figured out how to get things to load we should get some progress.

---8<---
Loading is TBH pretty awful. As suspected repeated 0x00 causes the tape reading to stop. I'm not sure how many, but I've sidestepped the issue by escaping all of them. And then escape character needs to be escaped as well. All of this lengthens the program but if it becomes a problem then I'll add compression as well.

zx70, 2025-10-07:
could the interrupt be rewritten/wrapped ?

suborb, 2025-10-07:
> could the interrupt be rewritten/wrapped ?

Nope, it's all in ROM. No way around it apart from using im2 (if it works on the hardware) or disabling interrupts.