z88dk/z88dk #841: (classic) Add Casio FP-1100 support

suborb, 2018-07-07:
Just investigating this machine.

* Z80 @ 4Mhz  - 64k RAM
* uPD7801G @ 2MHz - 48k VRAM - 3 planes of RGB ala Multi8 + 40/80 column
* Bank1: 0x000 -> 0x8fff, 0x9000->0xffff port 0xffa0?

SubCPU also handles keyboard (not sure how it's passed back as of yet)

Emulator: takeda, CP/M boot discs: http://www.retroarchive.org/maslin/disks/casio/index.html

Can boot from a disc - first sector is a bootstrap that can load the rest of the program. I've attached a disassembly of the CP/M boot sector. It's loaded to address 0x9000. Code at 0x0356 in BASIC.ROM loads a sector from disc. hl = address to load, c = track, b = sector. e = head?

Suspect $074f might be print character in a to screen.

[bootsec.txt](https://github.com/z88dk/z88dk/files/2173063/bootsec.txt)

There's some out(c) calls in here that probably setup the paging, yet to be investigated/matched up with the emulator source.

The emulator reads .td0 and then writes .d88 versions of the disc (so a nice converter!) when changes are made. It can also read d88.

d88 is a fairly simple format to include within appmake. Disc format is the usual CP/M one (http://www.cpm8680.com/cpmtools/cpm.htm) but if we're bootstrapping then we don't need a filesystem.

zx70, 2018-07-09:
I'd suggest for the SPC-1000, the MULTI 8 and eventual new targets to set the graphics attributes "reverted" (black over white) by default.    Is it feasible ?

suborb, 2018-07-09:
I think we need to create a set of functions that take a colour parameter(s). plot = ink, unplot = paper,  xor = ink + paper

Quite a few ports now theoretically support pixel perfect colour.

suborb, 2018-07-09:
We can boot hello world at least:

[image]

suborb, 2018-07-10:
Why did I pick this machine? It's not going to be a nice target - the VRAM access via the subcpu is very slow (64us a byte). It looks like the Vegetable Crush game that's mentioned (but I can't download since the links have expired) actually ran on the subcpu?

Some more info:

* http://d.hatena.ne.jp/tanzi/20150623/p2 
* Manuals (in French): http://casio.ledudu.com/pockets.asp?type=1805
* Spanish thread: http://retrowiki.es/viewtopic.php?f=76&t=200031679

tschak909, 2019-02-17:
daaaarn, really wanted to port platoterm to this :(

-Thom

suborb, 2019-02-18:
This machine was horrible to add support for!

If you can find out the subcpu commands then plotting can be done - I’m not sure I found any software running on the z80 side that did plotting though so I couldn’t disassemble.

puppet106, 2019-08-22:
If you need software or manuals for the Casio, I think I have it all.

This was my first computer, by the way.