z88dk/z88dk #845: Sharp MZ2500

zx70, 2018-07-09:
I think I have enough material to be able to support this monster.

zx70, 2018-07-10:
I'm a bit ashamed for the minimal support I achieved so far, by the way I could compile and run program with size <= 4K, basic console input/output only.

suborb, 2018-07-10:
Filesystem problems? What's the filesystem used?

zx70, 2018-07-10:
I sent you the information I have.   At the moment I'm experimenting trying to bend my simple appmake tool, but no luck so far.

zx70, 2018-07-10:
..    the code I committed is an attempt to boot the program directly from disk, hopint there are no size restrictions on the IPL program !

I was able to access the text memory and attributes in CP/M but the memory banks were probably arranged differently. .   the last experiment was this one:

main()
{
	
#asm

;MRSEL	Memory address	MRSEL	Memory address	
;0		0000h - 1FFFh	4		8000h - 9FFFh	
;1		2000h - 3FFFh	5		A000h - BFFFh	
;2		4000h - 5FFFh	6		C000h - DFFFh	
;3		6000h - 7FFFh	7		E000h - FFFFh

	ld	a,5	; A000h - BFFFh
	out ($b4),a

;MB			Memory block
;00h - 0Fh	Main RAM (8K blocks, 16 = 128K)
;10h - 1Fh	Extended RAM (8K blocks, 16 = 128 K)
;20h - 27h	Standard graphic VRAM (B / R / G / I)
;28h - 2Fh	Extended graphic VRAM (B / R / G / I)
;30h - 31h	Standard graphic VRAM (Read Modify Write)
;32h - 33h	Extended graphic VRAM (Read Modify Write)
;34h - 37h	IPL ROM	On M1 cycle
;38h		Text VRAM
;39h		Kanji ROM / PCG
;3Ah		Dictionary ROM
;3Ch - 3Fh	Communication ROM
	
; A800 = COLOR ATTRIBUTES
; B000 = FONT select

	ld	a,$38	; text ram
	out ($b5),a
	
	ld hl,$B000
	ld de,$B001
	ld bc,80*25-1
	ld (hl),139
	ldir

	ld hl,$A000
	xor a
loop:
	ld (hl),a
	inc a
	inc hl
	jr nz,loop
	di
	halt
	
	
	ld hl,$A000+80
	ld a,0
	ld (hl),a
	inc hl
	inc a
	ld (hl),a
	inc hl
	inc a
	ld (hl),a
	inc hl
	inc a
	ld (hl),a

	inc hl
	
	ld a,100
	ld (hl),a
	inc hl
	inc a
	ld (hl),a
	inc hl
	inc a
	ld (hl),a
	inc hl
	inc a
	ld (hl),a

	di
	halt

#endasm
	
}

zx70, 2018-07-11:
I'm slowly proceeding, I found a way to halve the disk image size and I'm experimenting with the bank switching and the disk format..  I just discovered that the only working entry for the user programs is 32768, even if I shift slightly the program in the same memory block, so some static setting must be left somewhere.
Current memory configuration:
#include <stdio.h>

int readbank(int b) 
{
	#asm
	ld	a,l
	out ($b4),a
	in a,($b5)
	ld h,0
	ld l,a
	#endasm
}

int i;

main()
{
	printf("Memory bank configuration\n");
	printf("-------------------------\n\n");
	for (i=0;i<=7;i++)
		printf("Bank %u, value %u\n",i,readbank(i));
}

[image]

zx70, 2018-07-11:
I think this memory organization is described directly in the boot disk image..   in appmake I put:

	writeword_xor(0x30,fpout);					/* "start sector"?, seems rather to be a relative offset to the "memory bank organization" below */

	writebyte_xor(12,fpout);					/* Memory bank: 9..15 for $2000..$E000 */
	//writebyte_xor(13,fpout);					/* next Memory bank ? */
	
	
	writebyte_xor(0xff,fpout);					/* load sequence termination ? */
	
	for	(i=1;i<=14;i++)
		writebyte_xor(0,fpout);

	/* memory bank organization at boot (Mapping at start of execution ) */
	writebyte_xor(8,fpout);		/* bank for $0000 */
	writebyte_xor(9,fpout);		/* bank for $2000 */
	writebyte_xor(10,fpout);	/* bank for $4000 */
	writebyte_xor(11,fpout);	/* bank for $6000 */
	writebyte_xor(12,fpout);	/* bank for $8000 */
	writebyte_xor(13,fpout);	/* bank for $A000 */
	writebyte_xor(14,fpout);	/* bank for $C000 */
	writebyte_xor(15,fpout);	/* bank for $E000 */

The first bank is changed by the z88dk startup code to support rst 18h etc.
By changing the "memory bank" byte I can move the program position on any bank between $2000 and $E000, I hope the next bytes (starting from the one I commented as $FF) will permit the boot loader to pick more blocks from the disk.  A simple extension of the "byte list" adding more banks didn't work, though.

Fabrizio-Caruso, 2018-09-12:
I have a positive report: I can run a turn-based version of my game (just stdlib).

In order to get my game working I have to reduce ram use to a minimum (by some options in my Makefile).
I looks like this target has access to very little ram. I am not talking about the disk size.
There seems to be a ram issue.

suborb, 2018-09-17:
After translating the SVC.INC file, I think an approach that will allow larger files to be loaded is:

We have a bootstrap program that then loads the file from disc. It looks like DEVFN/LOPEN or DEVNM/LOADF do what we need.

In order to do this we need to know:

- what format a OBJ/BTX file is.
- write a disc in MZ format

suborb, 2018-09-19:
I think the CP/M disc is a normal MZ disc, there's this IPL sector:

```
00001000  01 49 50 4c 50 52 4f 50  2d 43 50 2f 4d 20 4c 4f  |.IPLPROP-CP/M LO|
00001010  41 44 45 52 0d 00 00 00  00 80 00 00 00 00 80 00  |ADER............|
00001020  02 03 04 05 06 07 0b ff  00 00 00 00 00 00 00 00  |................|
00001030  00 08 09 0a 0b 0c 0d 00  00 00 00 00 00 00 00 00  |................|
00001040  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
```

So I suspect it's loading a bootstrap that contains enough code to implement the CP/M filesystem. This then loads PCPM.SYS to address 0xc000

Note the 0x80 at offset 0x101e rather than the 0x30 that we have in appmake. This presumably allows it to skip over the CP/M directory and into one of the two system files on disc.

The 80 column "Loading Personal CP/M" text that appears during booting is at offset 000153d0 (xor'd) on disc.

suborb, 2018-09-22:
I've now managed to load more than 4kb onto the MZ2500.

It looks like the IPL program is actually loaded from block $30 to block $90 so 24k. I've discovered the IPL call for loading sectors so we could load more if we want/need to. Thus, we could create a 64k target if desired and take over the entire machine.

I've got a bit of cleanup to do, but this should be in the build tonight. In the meantime, here's the Adventure A (about 16k) screenshot:

[image]

zx70, 2018-09-22:
so fast!   thank you for the great improvement!

suborb, 2018-09-23:
I've switched the default screen printer to gencon so we get some positioning and ink colour as well as a 40/80 column switch mode.

This is using the IPL screen printer - I've not managed to figure out how exactly an ASCII code is broken down into a PCG bank identifier and character index.

zx70, 2018-09-24:
clear, it is a complex architecture and it takes time to figure it out